MarlinFirmware/Marlin #9363: Bltouch does not detect the bed

Bernardo1996, 2018-01-26:
Hi guys,
I have recently configured my bltouch and I noticed that when I do the leveling of the plate happens at least once the nozzle slams on the plate and continues to go down, what is it?

Bernardo1996, 2018-01-26:
in addition to the problem before, this also happens to me.
https://youtu.be/EwDOSWrm-LU

Bob-the-Kuhn, 2018-01-27:
You may need a new pin/probe for your unit.  On deltas the downward force during a head crash is usually low enough that it doesn't damage the BLTouch.  On my printer it usually ruins the pin.

Contact the BLTouch people to see if they'll send you a new pin.

It might just need adjusting or cleaning.

To see if it needs adjusting/cleaning run the self test for several hours.  If it stops after only an hour or two then it isn't reliable enough for use on a printer (in my opinion).

To enter the self test mode issue **M280 P0 S120** and then the probe continually deploys and retracts.  If it stops before you reset the controller then it had a problem.

There's a setscrew on the top of the unit.  You'll need a 5/64" allen (hex) wrench to adjust it.  Mark your starting location and keep track of how much you've adjusted it so you can get back to the starting position if needed.

I've seen an adjusting procedure for the BLTouch but I can't find it right now.

To clean it remove the setscrew, turn it upside down so the pin falls out and clean the pin with alcohol.

Bernardo1996, 2018-01-27:
it seems strange to me because the bltouch is new, I did not understand what I have to do with that hexagonal nut

Bernardo1996, 2018-01-27:
this is what happens to me, in addition to the fact that the blotuch goes into alarm, it happens to me that the nozzle keeps going down and the bltouch does not send the signal to stop it.
https://www.youtube.com/watch?v=GDLeYqSkSZ8

Bernardo1996, 2018-01-27:
I followed this guide and now does not flash anymore, but continues to crash on the bed.
https://youtu.be/eDfGZ4bjxiM

Bernardo1996, 2018-01-27:
https://youtu.be/njW2fYqgw1w

Bob-the-Kuhn, 2018-01-28:
There's a servo probe subtest in the M43 command. In Configuration_adv.h enable PINS_DEBUGGING and then issue **M43 S**. After that follow the prompts on the host screen.  I'd like to know if the probe deploys and retracts properly and if it detects the 5mS pulse.

I just saw another BLTouch thread where enabling  ENDSTOP_INTERRUPTS_FEATURE fixed their problem.  I don't understand why it fixed the problem but it's worth a try.

Bernardo1996, 2018-01-28:
I heard that some people have increased one of these values ​​by one bit. Z_CLEARANCE_DEPLOY_PROBE and Z_CLEARANCE_BETWEEN_PROBES could both be increased by one bit (in Configuration.h).
However I have tried that changing the height of the sensor, the problem occurs less. maybe setting the right height and enabling this command you said should be fine...

Bernardo Carnevale.

ExtremeSitting, 2018-01-28:
I had similar nozzle crashing issues when I got mine installed as well. The problem in my case was the probe position wasn't perfect. There is a [document](https://docs.wixstatic.com/ugd/f5a1c8_57a23e0d76714fcab4508fdd3ae9ab50.pdf) that comes with the BLtouch that shows measurements in relation to the hotend and nozzle. They are not a suggestion. The probe has to be exactly with in those tolerances or it will crash. Once you have it physically positioned correctly, you have to measure it's real location in relation to the tip of the nozzle and put those offsets in the [marlin config](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration.h#L686-L688).

I've found that running a couple of tests after installation or adjustments to anything can save a lot of time with the BLtouch.

1. `M280 P0 S120` This is a self test that makes the probe deploy and stow in a loop. `M280 P0 S260` resets the probe. If it doesn't do anything when you start the self test you can start troubleshooting.
2. Lift the nozzle off the bed ~100mm. Home Z but touch the end of the probe with your finger. It should stop and lift ~4mm and go in for another touch. Let it hit your finger again. This shows that the probe is sensing as it should.
3. If you're still concerned the nozzle may crash, you can set a hollow test cube upside down under the probe location. That will give you a nice buffer to not let the nozzle crash while you're testing. It's cheaper to replace the probe than it is to replace the bed.

A final note: Don't remove the core. Doing so is only necessary IF you need to clean the pin, but that's unlikely since it's new. The same document that shows how to mount it comes with measurements on how deep the core should sit from the top of the probe body. You can adjust that if you find it's not with in spec. Ultimately, you should subscribe to the BLtouch google group if you need additional support for the probe.

Roxy-3D, 2018-01-29:
Is anybody seeing a double click of the pin like I noticed here:  https://github.com/MarlinFirmware/Marlin/issues/6807#issuecomment-361113464

Bernardo1996, 2018-01-29:
ok, I think I have solved the problem that the nozzle crashes in the bed (I had badly set the height of the bltouch).
now only the problem remains that the bltouch flashes and the bar extends and retracts quickly.
how do you solve this?
https://www.youtube.com/watch?v=aTHRSPOtcE8

ExtremeSitting, 2018-01-29:
I've never experienced that. However, two things come to mind: Bad probe or signal interference.

I'm leaning toward bad probe, because it seems like the solenoid is letting the pin go randomly immediately after a retract action. I own 2 probes that are mounted to interchangeable tool-heads and I've never had issues swapping things back and forth. The probes seem to be pretty tough for what they are and manage to maintain a high degree of accuracy when being moved around. Given the one you have is new, it may just be a manufacturing or even a calibration defect.

Check out that core screw I mentioned above. It is basically the stop that sets the pin in the right place relative to the solenoid. If the pin is not 'deep' enough (the core screw is too far in) it would let the pin go too soon, or it wouldn't be able to hold the pin while retracted. Go through the measurement section and check the depth on that screw.

I'm tempted to write off signal interference, because unless you're running the sense wires in the same bundle of wires running x or z movement it is very doubtful there's anything providing a source of cross-talk while the probe is working.

Do you have the Smart or Classic probe? Have you asked Antclabs directly about this issue?

Bernardo1996, 2018-01-29:
yes, all the threads of the bltouch are close to those of the movement x z. however, I adjusted the screw as written in the manual

Bernardo Carnevale.

ExtremeSitting, 2018-01-29:
I re-watched the video you provided and answered one of my questions, but I'm still not sure about the re-deploy issue. Does probing fail at points 4 and 7 every time?

If that is your printer then you have a classic. There's not a huge difference between the 2 versions, but the pin design did change. Setup seems to be roughly the same as well. I have the 'smart' version; FWIW.

I've compiled a list of notable config options based on the video:
1. [Enable probe with delay](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration.h#L640-L643). The BLtouch has received a lot of love in marlin in past releases, so simply enabling this gets you almost 100% of the available functionality. You should find a BLTouch menu item via your LCD screen where you can test, deploy, stow, and reset the probe.
2. [Set probe offset from nozzle](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration.h#L686-L688). I covered this one before.
3. [Deploy clearance](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration.h#L718-L719). This can be handy. The compact nature of the probe as well as it's deploy and stow accuracy mean you can drop these down pretty low. However, it's not necessary. I have mine set to to 5mm for both, but you may want to test a bit more vertical clearance.
4. [You shouldn't need to modify this.](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration.h#L721-L723) The probe has a very small deployed length relative to stowed and surely will never exceed the values here.
5. [Bilinear config settings.](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration.h#L910-L942) Based on the video, it looks like you're using bilinear bed leveling. I do too. The probing positions and the subdivisions are the only thing you should need to change and it looks like that's what you've done.
6. [Safe homing and location](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration.h#L1034-L1043). I enable safe homing because I need to know where the nozzle is to compensate for the location of the probe. I also home on the center of the bed. You don't film homing, but you should make sure you're using safe homing.

Optional: [Enable the repeatability test](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration.h#L725-L726). I doubt this will tell you anything useful for the issue you're facing, but it could tell you if the probe is lacking accuracy and suggest the probe is bad. I'm really not sure though.

That's basically everything you should need to set for your average printer. Preferred test points are really the only thing you'll end up changing after the initial config to get the probe working.

I'm sorry for the wall of text here, but I'm really hoping we can get your probe working.

Bob-the-Kuhn, 2018-01-29:
Sanity check - does the top of the pin look like the first or second picture?  If the top is squashed then you won't be able to get it to work reliably.  The pin must be replaced.

[image]

[image]

I think the next thing to do is to adjust the core (set screw) until the **M280 P0 S120** test runs reliably.  Start with the core flush to the top and then turn the hex wrench clockwise 1/2 turn at a time.  To clear the error state issue a **M280 P0 S60**.

Bernardo1996, 2018-01-29:
I think it's a bit casual, it does not happen only in points 4 and 7. however, changing position could help me in some way? (how you do it?) and the fact that the bltouch starts blinking as it can be solved?

ExtremeSitting, 2018-01-29:
@Bob-the-Kuhn The differences appear to be limited to a blue LED when testing, and a 3.3v logic conversion according to jumbled documentation. From my own experience, the pin has a slightly different shape as well. I think it's just to make the pin weigh less.
This is best picture I have of the pin without taking one of them apart.
[image]

@Bernardo1996 I don't think changing the location of the probe would help, but changing the clearance may. Once it probes the first time, have it raise the nozzle >5mm. I'm curious if it still has the same issue. Also, check the pin as @Bob-the-Kuhn mentioned. If you crashed the probe, then it could be mushroomed and needs replacement. I didn't account for that in my original response.

Bernardo1996, 2018-01-29:
thanks for the suggestions, however, the threads of the bltouch can not be close to those of the engine x and y?

ExtremeSitting, 2018-01-30:
I've seen it mentioned in other places as a potential issue (never verified as the cause). I've never personally experienced issues even though the sense wires and stepper wires are basically in the same unsheilded bundle. I only mentioned it as a possibility because the probe seems to deploy during an axis move instead of after or before an axis move. I think it's a better use of time/energy to check the pin and the integrity of the probe.

Bernardo1996, 2018-01-30:
Oook

closed 2018-02-14 by thinkyhead