MarlinFirmware/Marlin #9366: [Bugfix 2.0.x] Gcode parser broken since #9327

benlye, 2018-01-27:
Since PR #9327 the gcode parser appears to be ignoring some parameters.  For example, `M155 S2` no longer has any effect meaning OctoPrint hosts will automatically disconnect from an idle printer.

Reverting 80d3ded89523b741cec40c43cc09673411eaf168 and 399bca316a5abe64f18ce202699822fcd9738b98 seems to 'fix' it and the parser works again.

Also, trying to compile with `FASTER_GCODE_PARSER` disabled fails:
```
In file included from Marlin\src\gcode/gcode.h:250:0,
from Marlin\src\Marlin.cpp:49:
Marlin\src\gcode/parser.h: In static member function 'static bool GCodeParser::seen(char)':
Marlin\src\gcode/parser.h:147:43: error: 'valid_float' was not declared in this scope
if (b) value_ptr = valid_float(&p[1]) ? &p[1] : (char*)NULL;
^
```

#9353, #9350

gavatron3000, 2018-01-28:
is the issue comparable to mine? manually jog printer with lcd no worries. but issue a jog on any axis from octoprint nothing happens and i get a disconnect? (re-arm)
it will G28 and G29 from octoprint

benlye, 2018-01-30:
#9393 should fix this.

gavatron3000, 2018-01-30:
Fingers crossed

Bob-the-Kuhn, 2018-02-01:
Assuming that you're compiling for 8 bit boards, yes, the gcode parser is broken.  Any parameter/argument that uses "P" and above is ignored.

PR #9393 will fix it once we're sure that we're not breaking something else.

gavatron3000, 2018-02-01:
I'm having issues on the re-arm with any movement gcode that's feed through serial not working. G28 G29 both work and moving axes from the LCD work. But try to jog through octoprint or start a print it just disconnects and runs through the gcode without moving

benlye, 2018-02-01:
> Assuming that you're compiling for 8 bit boards, yes, the gcode parser is broken. Any parameter/argument that uses "P" and above is ignored.

Yep, RAMPS on a mega 2560,  compiling  in platform io. I've been following the discussion.

closed 2018-02-14 by thinkyhead