MarlinFirmware/Marlin #9367: Crash compile bugfix 2.0.x on  #define CNC_COORDINATE_SYSTEMS

cz3dtc, 2018-01-27:
on uncomment #define CNC_COORDINATE_SYSTEMS  in  config h adv

In file included from sketch\src\Marlin.cpp:49:0:

sketch\src\gcode/gcode.h:400:10: error: 'bool GcodeSuite::select_coordinate_system(int8_t)' cannot be overloaded

     bool select_coordinate_system(const int8_t _new);

          ^

sketch\src\gcode/gcode.h:278:17: error: with 'static bool GcodeSuite::select_coordinate_system(int8_t)'

     static bool select_coordinate_system(const int8_t _new);

                 ^

exit status 1
Error compile Arduino/Genuino Mega or Mega 2560.

GMagician, 2018-01-28:
Fixed

cz3dtc, 2018-01-31:
Marlin-bugfix-2.0.x config crash 

`//======================================================================
/**************** TEVO TARANTULA EASY CONFIG ***************************
            Original idea by terryb.print3d@gmail.com
             Modified by jb.github@rcairgallery.com

    The latest version of Terry's original file will always be found at:
          https://github.com/terryb58/Marlin-EasyConfig

    The latest version of this file (and complete firmware) will always
  be found at:
          https://github.com/JimBrown/MarlinTarantula

    This is an attempt to create a simple configuration for as many
  different Tevo Tarantula variants as possible.  This will always be
  a work in progress. Email me if you have any questions, suggestions,
  or if you encounter problems when using Easy Config.

    This is a Marlin 2.0.x configuration file. I will update this as
  new versions of Marlin are released.

    NOTE: Sanity check should still work and should not show any errors.
      Please report any errors.  Thank you.

    NOTE: Don't forget to do an M502 followed by an M500 any time you
      upload the firmware.

    See https://youtu.be/-sQ8p00pG5E for an excellent tutorial on using
    this firmware.

***********************************************************************/

/**
 * Equipment options
 */
#define LARGE_BED
#define SDSUPPORT
//#define CHANGE_Y_DIRECTION      // If your bed homes in the wrong direction front to back, enable this.
//#define CHANGE_X_DIRECTION      // If your X carriage homes in the wrong direction left to right, enable this.
//#define CHANGE_Z_DIRECTION      // If your Z homes in the wrong direction bottom to top, enable this.
//#define HOTEND_E3DV6            // Genuine E3D v6 hotend.
//#define FULL_GRAPHIC_SMART      // Enable this if you have a RepRap Discount Full Graphic Smart Controller (The
                                  // stock controller is a RepRap Discount Smart Controller)
//#define Z_DUAL_STEPPER_DRIVERS  // Enable this if you have dual Z stepper motors with the second stepper motor
                                  // connected to the next available E plug (usually E1)

/**
 * Offset from endpoints to get nozzle to 0,0 (front/left of bed)
 * (How to center prints: https://github.com/JimBrown/MarlinTarantula/wiki/How-to-center-your-prints-(EasyConfig))
 */
#define NOZZLE_X          0
#define NOZZLE_Y          0

/**
 * Minimal Z height (in mm) before homing (G28) for Z clearance above the bed, clamps, ...
 */
#define Z_HOMING_HEIGHT 5

/**
 * Primary Extruder steps per mm (plugged in to E0 port on controller)
 * (How to calibrate: https://toms3d.org/2014/04/06/3d-printing-guides-calibrating-your-extruder/)
 */
#define E0_STEPS      400 // Stock extruder. If you have a Tevo Titan, try 400 then calibrate.
#define CHANGE_E0_DIRECTION   // If your extruder is going backwards, enable this.

/**
 * Z Axis steps per mm (Default for stock lead screw is 1600)
 * If you install a lead screw with a different pitch and/or lead, change this
 */
[... 3650 lines cut ...]
   */
  #define MAX7219_DEBUG_PRINTER_ALIVE    // Blink corner LED of 8x8 matrix to show that the firmware is functioning
  #define MAX7219_DEBUG_STEPPER_HEAD  3  // Show the stepper queue head position on this and the next LED matrix row
  #define MAX7219_DEBUG_STEPPER_TAIL  5  // Show the stepper queue tail position on this and the next LED matrix row

  #define MAX7219_DEBUG_STEPPER_QUEUE 0  // Show the current stepper queue depth on this and the next LED matrix row
                                         // If you experience stuttering, reboots, etc. this option can reveal how
                                         // tweaks made to the configuration are affecting the printer in real-time.
#endif

/**
 * NanoDLP Sync support
 *
 * Add support for Synchronized Z moves when using with NanoDLP. G0/G1 axis moves will output "Z_move_comp"
 * string to enable synchronization with DLP projector exposure. This change will allow to use
 * [[WaitForDoneMessage]] instead of populating your gcode with M400 commands
 */
//#define NANODLP_Z_SYNC
#if ENABLED(NANODLP_Z_SYNC)
  //#define NANODLP_ALL_AXIS  // Enables "Z_move_comp" output on any axis move.
                              // Default behaviour is limited to Z axis only.
#endif

#endif // CONFIGURATION_ADV_H
`

closed 2018-02-14 by thinkyhead