MarlinFirmware/Marlin #9368: TMC2130 - Printing issues, Shifting Layers

eLement87, 2018-01-27:
Hello togehter,

**tl;dr**
I had problems to get the TMC2130 working.
When the Printer is moving in Example the X-Axis from left to right - 0mm to 10mm - than every Step back to 0 cause a shift to right. 0 - 10 - 3 - 15 - 6 - 19 and so on. I don't know if the Steppers are losing steps?! This happends on every Axis. Sorry it's hard to explain in englisch ;)  One ore more Layers are okay and then it produce a extremly Layershift. The drivers are getting realy hot even if you drive only 30mm. (Heatsink on every driver and active fan does the job)

[image]

**First Things first**

Printer: Tevo Tarantula
Stepper: Tevo Nema 17 17HD4401-C5.18 1.8° (maybe [Busheng 17HD40005-22B, 36Ncm, 1.3A, 2V](https://github.com/superjamie/lazyweb/wiki/3D-Printing-Stepper-Motors-and-Drivers)) 
Driver: 4x TMC2130 from Watterott
Mainboard: MKS 1.4 from Amazon
Marlin: 1.1.x (1.1.x Bugfix Branch from 25.01.2018)
TMC2130 Library: teemuatlut/TMC2130Stepper 2.1.1

Since i bought the Printer i used the Original Board from Tevo with integrated driver with flashed Marlin. Nevermind in January i watched the YT Video from Tom with the SilentStepStick. So i got myself four TMC2130 Boards from Watterott. In order to get the things working i soldered the Boards and do the requiered stuff in Marlin:

[image]

**Configuration.h**
```
#define BOARD_MKS_13
//                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 1600, 100 }
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 7, 50 }
#define DEFAULT_MAX_ACCELERATION      { 3000, 3000, 100, 10000 }
```

**Configuration_adv.h**
```
#define HAVE_TMC2130

#define X_IS_TMC2130
#define Y_IS_TMC2130
#define Z_IS_TMC2130
#define E0_IS_TMC2130

#define R_SENSE           0.11  // R_sense resistor for SilentStepStick2130
#define HOLD_MULTIPLIER    0.5  // Scales down the holding current from run current
#define INTERPOLATE       true  // Interpolate X/Y/Z_MICROSTEPS to 256

#define X_CURRENT          800  // rms current in mA. Multiply by 1.41 for peak current.
#define X_MICROSTEPS        16  // 0..256

#define Y_CURRENT          800
#define Y_MICROSTEPS        16

#define Z_CURRENT          800
#define Z_MICROSTEPS        16

#define E0_CURRENT         800
#define E0_MICROSTEPS       16
```

**pins_RAMPS.h** - Changed the Pinmapping for the LCD
```
#define X_CS_PIN           44 // 53 -> Pin D44
#define Y_CS_PIN           66 // 49 -> Pin D66
```
All Jumpers beneath the drivers are set.

As i mentiond above the Printer / Axis is not going back to his starting Point. Every Step back (to the left) it gets away from 0. There are several Posts with similar behaviour - #8478 #8967 #8927 
But i have no shorts to ground nor a Chinese SilentStepStick. I changed the Current, tighten the Belt and check everything mentioned in the other issues.

Maybe the Mainboard is faulty or the Steppers? I think it's not a mechanical problem, but... i don't know!?
With the Stock Mainboard i can't notice this behaviour.
Is there something that i forgot?

Thanks and best regards
Ben

teemuatlut, 2018-01-27:
Lower your acceleration and max speed settings and see if that fixes anything.
Start with say 1200mm/s^2 and 80mm/s. The numbers are lower than need be but it's better to over compensate and go start increasing from there.
StealthChop or SpreadCycle?

gavatron3000, 2018-01-28:
these get super hot, your running heatsinks right?

franji, 2018-01-28:
you could try to install marlin 1.1.6 and use 650mah. I have the same problem with my tmc2130 when I use a version higher than 1.1.6.

oakdesign, 2018-01-29:
For any current over 500mA both heatsinks and active cooling should be applied

hpkevertje, 2018-01-30:
I had similar issues (along with z rising to infinity - #6581 and Y not moving on first home - #9018).
I managed to get 1.1.8 working stable by disabling stealthchop and using only spreadcycle.

That did cause issues with sensorless homing. I applied what maniacgit describes in #8890. 

After all this I have a working system and no more layer shift.
I am thinking about going back to 1.1.6 to be able to use the drivers in stealthchop mode.

eLement87, 2018-01-31:
Sorry for the delay. Thanks for the answers!

@teemuatlut
StealthChop 

Setting the acceleration to a lower value did the trick, i think. I set the default acceleration to 800 and Max to 1000. Values over 1200mm/s shows the same result.
```
#define DEFAULT_MAX_ACCELERATION      { 1000,  1000, 70, 10000 }

#define DEFAULT_ACCELERATION          800    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  1000    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   1000    // X, Y, Z acceleration for travel (non printing) moves
```
[image]

Not a optimal print but there are no more Layershiftings! 
Now it moves +- 0 - 10 mm forwards and back. But i have to investigate a bit more.

I installed heatsinks on every driver and a 120mm Fan on over the Mainboard.
I will be testing on weekend.

ikarisan, 2018-01-31:
I have the same / a similar problem with my TMC2130 drivers  (Watterot with heatsinks) on X and Y and Marlin 1.1.8 (bugfix-1.1.x). Small object are printed very smooth and without any layer shifts. But if I print larger objects (https://www.thingiverse.com/thing:2484766) I print them for the bin. :( 

[image]
[image]

With my Anet AM8 I am printing at 90mm/s and a travel speed of 130mm/s, using these values:

`
[...]
  #define DEFAULT_MAX_ACCELERATION      { 2000, 2000, 100, 10000 }
  #define DEFAULT_ACCELERATION          400   
  #define DEFAULT_RETRACT_ACCELERATION  1000    
  #define DEFAULT_TRAVEL_ACCELERATION   1000  
  #define DEFAULT_XJERK                 15.0
  #define DEFAULT_YJERK                 15.0
[...]
`
`
[...]
  #define X_CURRENT          700 
  #define X_MICROSTEPS        16
  #define Y_CURRENT          700
  #define Y_MICROSTEPS        16
  #define STEALTHCHOP
  #define HYBRID_THRESHOLD 
  #define X_HYBRID_THRESHOLD     100  // [mm/s]
  #define Y_HYBRID_THRESHOLD     100
[...]
`
I used the same values for acceleration, jerk and printing speed on my old stock Anet A8 board without any problems. :( 

I will try to print the same object with STEALTHCHOP disabled or using 1 as hybrid threshold.

teemuatlut, 2018-01-31:
It's always a balance of acceleration, driver current and driver temperature.
If you want higher acceleration, you need more current, which in turn requires more cooling.
If you can't provide enough cooling for the current, you need to lower acceleration.

You may want to tune max acceleration instead of default acceleration.

ikarisan, 2018-01-31:
Okay... with "hybrid threshold" set to 1 the result ist good! O.o

NeelD87, 2018-02-08:
@ikarisan can you please provide your config file. I am also using them on AM8 and having similar issues like you steps skipping on Y-axis.  Only works if I lower the speed to 45mm/s

teemuatlut, 2018-02-17:
@thinkyhead 
> We're finding that TMC drivers are prone to heat issues, but have been able to resolve them for the most part with bigger sinks and active cooling. What in your experience are reasonable ranges for current (and torque / acceleration / speed) when using StealthChop versus SpreadCycle modes?

Personal setup:
I_rms: X1150 Y1100 Z800 E850 [mA]
Max a: X2200 Y1800 [mm/s^2]
Max v: X120 Y120 Z8 E25 [mm/s]
I do have an 80mm Noctua fan cooling them and heatsinks on both sides of the X/Y chips. The Watterott protectors give some room in between making it possible.

I've never ran the drivers without active cooling outside of testing but I'd guess that with just heatsinks you'd be limited to below 850mA. Without heatsinks you may have to go as low as 650mA.
You can try reducing the PWM frequency (`stepperX.stealth_freq(0)`) so reduce losses.

Also should mention that stealthChop is meant for low/medium speeds. It's a bit harder to dig out what exactly those speeds are or if we're operating at speeds above intended.

I'm also hoping to one day better optimize the spreadCycle parameters and that coolStep could be used. This would allow the drivers to adjust their current according to the motor load and operate with a higher current ceiling as it would only be used for very brief durations. I've uploaded an updated [stallGuard example](https://github.com/teemuatlut/TMC2130Stepper/blob/master/examples/StallGuard/StallGuard.ino) to demonstrate how it works. I'd suggest using the IDE graph plotter.

ikarisan, 2018-02-17:
@NeelD87 Here they are....

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1734021/Marlin.zip)

sballano, 2018-02-17:
I just wanted to comment that I was having problems with my TMC2130 after my last firmware update. I updated from 1.1.6 to 1.1.8 and to the last TMC libraries. For some reason I got double the amount of movement as before (same steps/mm) on the X and Y axis, but not in the Z! and also more noise and missed steps even with higher current.

I reverted to the TMC library version https://github.com/teemuatlut/TMC2130Stepper/commit/58f01ad9e2dad9c36b784beab29885257078ff20 and now everything seems fine and I'm still run Marlin 1.1.8

teemuatlut, 2018-02-17:
Try the newest bugfix-1.1.x branch. There was a bug that was fixed a couple of weeks ago that may have been a likely cause for that.

sballano, 2018-02-17:
@teemuatlut unfortunately the bugfix branch didn't solve the double amount of movement with the latest TMC library version, I had to go back again to the one I was using

teemuatlut, 2018-02-17:
Did you powercycle the printer after upgrading?

sballano, 2018-02-18:
Yes. I program the board without the power supply, then disconnect the USB and connect the power supply after that

NeelD87, 2018-02-19:
@teemuatlut  does Prusa https://ultimachine.com/products/einsy-rambo-1-1 board use any passive or active cooling?

teemuatlut, 2018-02-19:
I don't really know. You'd have to go through the build manual to find out.
The board does have the advantage of a large ground plane that can dissipate more heat than the small step sticks. The drivers are also hardware limited to 960mA.
I can use just heat sinks with the Archim2 board which is the 32bit big brother of Einsy.

NeelD87, 2018-02-20:
seems accl is limited and current is also set to 400mA in software
SILENT_MAX_ACCEL  960

wbarber69, 2018-03-16:
I’m also running into this same issue with fystek drivers.  I’ve reduced all my current down to 500ma and I’ve reduced my velocity and acceleration to what teemault recommended and while the printers were progressing further and with less skips. It’s still happening.  Only now after it gets off for about 10 layers it rights itself.  So obviously acceleration is a big part of it.  I haven’t tried reverting to an earlier version of the libraries, but that is my next step.  I also haven’t tried running in spreadcycle yet to avoid doing too many firmware changes if it is a hardware issue. But obviously I didn’t have these issues until I installed the trinamic drivers.

wbarber69, 2018-03-17:
I’ve found a somewhat reasonable solution for this.  I have turned off stepper interpolation and am now driving the steppers at 16th stepping.  So far no layer skips.  My y motor is still a little warm so I need to tune my voltages a bit more but everything seems to a bit happier. One thing I will mention though.  If you have an octoprint setup and after interpolation is turned off, and you try to move the motors from the webapp you will notice a very loud grinding noise as you try to move the axes around.  Go into octoprint settings and turn down the feed rate for all your axes from 6000 down to around 2200mm/m.  I’m able to stay in stealthchop this way.

thinkyhead, 2018-03-18:
Thanks @wbarber69 — Hopefully from all the feedback we'll be able to add more information to the troubleshooting section on our [Trinamic Drivers page](http://marlinfw.org/docs/features/tmc_drivers.html).

wbarber69, 2018-03-18:
I would like to get slightly faster movements... but so far with interpolation turned off, I am printing.  300 acceleration, 120velocity, 7 jerk...  it could be better but no more layer shifting.  I may try turning interpolation back on but lowering stepping to quarter stepping.  After I finally get these parts finished so I can swap out my control board.  I have active cooling but right now it’s messy and not closed up right.  The part that’s frustrating is it was printing so f’n good before it skipped a layer... then it even righted itself and continued to print like it never skipped, then a few hours later it skipped again.

Recently the trinamic libraries were updated and I’m currently downloading the latest latest bugfix that’s been updated too.  Maybe something is better... I’ll see in a couple days.

thinkyhead, 2018-03-18:
You might try upping the voltage in your power supply. Most boards can handle up to 30V. But don't increase it too much if your board isn't rated for that. 14V shouldn't be a problem. Note that heat increases exponentially as current increases linearly.

wbarber69, 2018-03-18:
im already at 13.7v but ill give it a shot

Lyr3x, 2018-03-18:
@teemuatlut What kind of heatsinks do you use for the down side? On top i'm using the 9x9x12.

teemuatlut, 2018-03-18:
[8.8x8.8x5](https://www.ebay.com/itm/10pcs-Aluminium-8-8x8-8x5mm-Heatsink-Heat-Sink-Diffuse-Cooling-Cooler-Fin-Black-/311604946150?hash=item488d1a24e6). I think those would be close.
But you of course need the StepStick protector boards to create some space.

Lyr3x, 2018-03-18:
This one: http://www.watterott.com/de/SilentStepStick-Protector ? Should i use the board as well? Or just the bigger sockets?

thinkyhead, 2018-03-19:
With the protector it looks like the unit would become _crazy tall_ once you flip the pins for the SPI connection.

teemuatlut, 2018-03-19:
@Lyr3x Yes, those are the ones. If you buy the actual boards of course you should also use them. You can get the taller headers for much much cheaper.

Lyr3x, 2018-03-19:
I'll buy those and test how much more current is possible... For me with Just heatsinks on one side about 820mA is Max. Speeds over 70mm/s results in layer shifts in stealthchop... Hybrid is working fine for me with your latest lib and the Bugfix branch(From Yesterday)

wbarber69, 2018-03-19:
Kai, you’re lucky I’ve barely been able to get past 500ma and 50mm/s.  I’m reconfiguring my cooling setup and installing some more heat sinks as well, hopefully by this weekend I should be back to interpolated steps.

Lyr3x, 2018-03-28:
With just one heatsink ontop my tmc's running fine @825mA (XY) for hours of printing... With reduced jerk (10) im able to print safely with 80mm/s and 150mm/s travel speed. Printed a very large object yesterday without any problems. 

I will install the second heatsink by this weekend. Hopefully i can push the current a bit more to push the speed to the limits 👍 💯

closed 2018-03-28 by thinkyhead

hapklaar, 2018-04-09:
I have this same issue, with just 30mm/s. Am I right in concluding this is caused by overheating?

teemuatlut, 2018-04-09:
Hard to say with the limited information you give. We need to know a lot more about your configuration and symptoms first.

wbarber69, 2018-04-09:
Printing speed as set in your slicer isn’t necessarily the issue.  You will need to reduce your velocities down from stock and max acceleration can come down as well.  I can print perfectly fine at up to 60mm/s (haven’t tried any faster) with my x and y velocity set 140.

Lyr3x, 2018-04-10:
As @teemuatlut said, we need much more information about your setup.

If you got your setup right, you even dont need to reduce acceleration or jerk to get some resonable speeds.

At the beginning i had some issues with the TMC's as well but with my current setup its just working perfect and silent:

1. Be sure that your Stepper Motors can handle up to 1.7A. Cheaper 0.9A steppers getting very hot (i mean very hot...). While overheating you are loosing much torque resulting in layer shifts.
2. To run the TMC2130's in stealhChop mode, i think that you need at least 875mA to achieve ~60-75mm/s without layer shifts
3. It would be much better if you can go crazy up to 1100mA or 1150mA for max speeds.
3.1 You definitly need active cooling (I'm using a 120mm Noctua Fan)
3.2 Double heatsink is adviced (see a few posts above)
4. Be sure you dont have any mechanical issues!

When you can achieve all points, you dont need do reduce your acceleration, jerk, velocity etc. to a absolute minimum which results in slower speeds.

With this setup i can print complicated and also large prints super stable @ 80-120 mm/s

wbarber69, 2018-04-10:
I can second that, kai.  I am currently running 1.2a steppers and they are just barely doing it.  I also switched over to titan extruders at the same time that require a much slower feedrate than what I’m used to so when I also turned down the velocity of of x and y it really slowed printing down.  So now I am facing start stop blobbing.  The next step is definitely higher rated steppers.  Then eventually it’s all going to 24v but until then I am happy with the quality.

hapklaar, 2018-04-10:
First thanks for responding so promptly. I was assuming the issue was universal, so didn't include any info about my setup, sorry about that.

I have an Anycubic Kossel built from a kit,  using Marlin 1.1.8 release as firmware. With the original A4988 drivers, printing worked fine, albeit very noisy. I bought some cheap TMC2130 clones from China and installed them without much issue. I changed settings in configuration_adv.h to enable the new drivers. At the moment I'm just using the included heatsinks and passive cooling and default settings in configuration_adv.h.

Printing was almost silent and seemed to go very well, until halfway into the printing process. It must have been somewhere around the 20th layer, I noticed X was starting shift to the right, every new layer shifted a bit more. I'm not sure what kind of stepper motors are used on this printer, but if I remember correctly they weren't very hot when the issue occurred, neither were the drivers. I'll be sure to check this next time it happens.

I've attached my config files, hope that's helpful. In the meantime I've downloaded the bugfix-1.1.x branch and copied my custom settings, so will try that one next weekend.

I would like to achieve a stable printing setup in StealthChop preferably without active cooling of the drivers. 30-45mm/sec printing speeds are totally acceptable for me with this printer.

[Marlin-manualbedlevelling-tmc2130.zip](https://github.com/MarlinFirmware/Marlin/files/1895228/Marlin-manualbedlevelling-tmc2130.zip)

wbarber69, 2018-04-10:
I wouldn’t recommend no active cooling.  I’m only using 2 40mm fans flowing across mine and it’s plenty of airflow.  What’s more important is that the air can move easily across the stepsticks which is kind of prevented by the header sticking up right next to the heatsink.  So I bent mine as you can see on the pictureS. Very easy to do, I just removed the plastic around the pins and used a sacrificial 4 pin DuPont connector to bend them 90degrees.

teemuatlut, 2018-04-10:
The clones have slightly worse cooling as even if the newer versions do have the intended thermal vias, they are still covered by solder mask. In the Watterott ones you have direct copper contact with the heatsink.
You can try running without a active cooling but this means that you also need to use lower current values and thus also lower acceleration values.
You can also try disabling interpolation. There was an interesting discussion at the [raise3d forums](http://forum.raise3d.com/viewtopic.php?t=6013) where a user solved his shifted layers issue this way.
Definitely should use the bugfix branches over 1.1.8. There was a critical bug that was fixed after the release.

wbarber69, 2018-04-22:
Damn it and I was just about to flash a firmware commit from just around that time.  Good thing I labeled all my changes this time...

#10446<https://github.com/MarlinFirmware/Marlin/issues/10446>

wbarber69, 2018-04-23:
> @wbarber69<https://github.com/wbarber69> — Most users don't see any loss of steps. It may be that the latest version of Cura is also involved in the recent issues some are seeing.

Perhaps but I may just hold out for a couple more days until some people have tested the reverted version.  I’m currently stable with a version that’s about a month or so old. I am upgrading my whole system to 24v and I was going to use that time to update the firmware, as my libraries are a little stale.

hapklaar, 2018-04-23:
@thinkyhead I use Cura 15.04 and have the same shift issues.

thinkyhead, 2018-04-24:
Cura's versions confuse me. I have 3.1. That sounds like it would be much older than 15.04.

Lyr3x, 2018-04-24:
Acutally 3.1 is much much newer. They are weird about Version Numbers :D... Since Cura 3.x i had a Lot Problems with predefined settings. I moved to slic3r since then

thjubeck, 2018-04-28:
This issue also exist with simplify 3D. Currently using a prusa i3 clone with simplify 3d and active cooling on the steppers with the skipping showing after a few layers.

Lyr3x, 2018-04-28:
What Kind of steppers? Which settings? (Accel., Jerk, etc)

thjubeck, 2018-04-28:
The Printer is an anet style straight from china with no labels on steppers below are the settings I am using. 

 */
#define DEFAULT_AXIS_STEPS_PER_UNIT { 100, 100, 1600, 90 }   // Upgraded to DRV8825 stepper drivers 6/16/17 and had to change this setting    REMEMBER TO CLEAR EEPROM!!!!    { 100, 100, 400, 90 } 

/**
 * Default Max Feed Rate (mm/s)
 * Override with M203
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 5, 25 }

/**
 * Default Max Acceleration (change/s) change = mm/s
 * (Maximum start speed for accelerated moves)
 * Override with M201
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_MAX_ACCELERATION      { 3000, 3000, 100, 10000 }

/**
 * Default Acceleration (change/s) change = mm/s
 * Override with M204
 *
 *   M204 P    Acceleration
 *   M204 R    Retract Acceleration
 *   M204 T    Travel Acceleration
 */
#define DEFAULT_ACCELERATION          3000    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  3000    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   3000    // X, Y, Z acceleration for travel (non printing) moves

/**
 * Default Jerk (mm/s)
 * Override with M205 X Y Z E
 *
 * "Jerk" specifies the minimum speed change that requires acceleration.
 * When changing speed and direction, if the difference is less than the
 * value set here, it may happen instantaneously.
 */
#define DEFAULT_XJERK                 10.0
#define DEFAULT_YJERK                 10.0
#define DEFAULT_ZJERK                  0.3
#define DEFAULT_EJERK                  5.0

Lyr3x, 2018-04-28:
Lower your Acceleration please to about 1000. Also you can try to set your x/y jerk to 12 or 15. After that you should check again if everything works fine

wbarber69, 2018-04-28:
I see a lot of info popping up here and there about the correct size and current motors playing a critical role in regards to tmc drivers.  What would be the recommended steppers to use with tmc2130 drivers?

Lyr3x, 2018-04-28:
So in my experience:
- Stepper Motors rated at least at 1.7A (1100mA RMS)
- Personally im running 17HS4401

Running my Maschine @3000 acceleration and Up to 200mm/s. Dont have any problems

Lyr3x, 2018-04-28:
The biggest Problem i had was really the steppers. While they were rated at .9A they overheated and lost torque

Artem-B, 2018-04-29:
This prompted me to look up the specs for the motors I use on my CR-10 w/ 2208.
I was somewhat shocked to see how quickly torque drops when the motors are driven by 12V.
[image]

AFAICT at 12V the motor would lose most of its torque at speeds > ~150 RPM -> 2.5 rot/sec -> 100mm/s (20-tooth GT2 pulley).

Higher-amp steppers fare quite a bit better and keep reasonable torque at much higher speeds, which backs up @Lyr3x 's recommendation:
[image]

Lyr3x, 2018-04-29:
Thank you for your Tests! This is very helpful :+1:

wbarber69, 2018-05-01:
> Yes that will work but I was looking for a solution not something to mask the problem.The settings have been the same through 3 different types of steppers (I like trying new things so always try the newest and latest things that come out and this was looking stable so I decided to switch). The other steppers worked fine with these settings.
>
> The actual fix was to disable interpolation and STEALTHCHOP which allowed to print at the same acceleration. I think STEALTHCHOP could be turned back on in hybrid mode but haven't tried that yet.

I recently upgraded my power supply and fans and heaters to 24v as I was informed that my whine during spreadcycle was due to low voltage.  But the motors still whine, anything else to look for?

teemuatlut, 2018-05-01:
You would need to do parametrization of spreadCycle to suit your motors.
I try to provide _some_ defaults that work but they're likely not optimal and I haven't done more comprehensive look into them when running pure spreadCycle, which I don't personally do. Finding better defaults might be something I could ask Trinamic to help with.
However, deciding on suitable chopper parameters is not an easy task as people can have a wide variety of motors with widely different specifications and users don't even run on the same voltage levels. This is not a problem for manufacturers like Prusa who only need the parameters to work with their own hardware where the specs are well known.
The spreadCycle parameters to look for would be off time, blank time, hysteresis values (start and end), current values. Possibly some others that I can't think of right now.

Trinamic's application note 001 provides more detail into the parameterization process.

wbarber69, 2018-05-01:
I’m using hybrid mode I should have said. On home I get whining maybe I need to look at something else

patternseek, 2018-10-10:
I noticed that the last Cura version was generating gcode with travel moves significantly over the speed limits I set in the settings. That's not 'the issue' exactly but it's worth noting that it happens. I hadn't had layer shifts previously (lightly upgraded A8) and then recently started getting them. That's what actually set me on the path that lead to an Sbase board and TMC2130s. So I was a little sad to end up with the same problem after replacing most of the printer! Anyway as others have noted, finding a balance between reducing accel, increasing current and cooling and having decent motors all helped.

[4 more replies]