MarlinFirmware/Marlin #9369: Homing on Max does not Behave same like on Min enstops

Yianni-VJ, 2018-01-27:
### Bug Report
- Description: Homing on Max does not Behave same like on Min enstops
- Expected behavior: My setup uses homing on the Max of X and Z axis. ****The expected behavior is that if the enstop is triggered before homing, the motor will move in reverse to disengage and then do the approach**. This only happens with enstops positioned on X Min and Z Min but not on Max.**
- Actual behaviour: Using Max endstops for homing does work but the #define X_HOME_BUMP_MM feature is not functioning, and if enstops are triggered in advance the position is set without accuracy.
- Steps to reproduce: Use my configuration files and attempt to home X on the Max end of travel. Also, manually trigger the X Max enstop and issue a G28 X.
[Configuration.h & Configuration.adv..zip](https://github.com/MarlinFirmware/Marlin/files/1670115/Configuration.h.Configuration.adv.zip)

thinkyhead, 2018-02-14:
Whenever there are homing or leveling issues, we ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark.

- Download Marlin `bugfix-1.1.x` to test with the latest "nightly" code.
- Enable `DEBUG_LEVELING_FEATURE` and re-flash the firmware.
- Connect to your printer from host software such as Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.

Examine the log output for the output of "Home 1 Fast:", "Move Away:", and "Home 2 Slow:" and let us know what it says.

boelle, 2019-02-20:
@Yianni-VJ did you try what thinky suggested? remember latest is now bugfix 2.0

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-06 by shitcreek