MarlinFirmware/Marlin #9370: number of steps rotated not the same for +1mm and -1mm

Namakemonoism, 2018-01-27:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1670134/Marlin.zip)

------------------------------------------------------------
### Bug Report
- Description: ---
Marlin SHORT_BUILD_VERSION "1.1.8"
Ramps 1.4 
A4988 on full stops
Printrun 1.6.0
steps/mm 11
This problem is not axis specific, both x and y experience it. 8825 driver also had the same issue, and a total of 3 motors were tested. Motor is not connected to an axis for testing. drivers were rand at 0.3v, 0.6v, 0.8v and 1v for tests.

When asked to move -1mm multiple times every other time rotates visually less degrees than expected. -2mm is ok. positive 1mm is also ok.

This problem happens with steps/mm set to 11.
Steps/mm set to 10 is ok.

- Expected behaviour: ---
+1mm and -1mm should move the stepper motor the same number of degrees each time.

- Actual behaviour: ---
+1mm moves a# steps, a# steps a# steps
-1mm moves a# steps, b# steps, a# steps, b# steps

- Steps to reproduce:
  - setup hardware like this:
          Marlin SHORT_BUILD_VERSION "1.1.8"
          Ramps 1.4 
          A4988 on full stops
          Printrun 1.6.0
          steps/mm 11
 -Connect by USB to Arduino 2560 in Pronterface
 -Send G91
 -Send G0 Y-1.0 multiple times
 - the number of steps turned should vary every other time.
 -send M92 Y10.0
 -Send G0 Y-1.0 multiple times
 - the number of steps turned should be the same each time.

Am I doing something wrong or break my config or can someone confirm this?

Namakemonoism, 2018-01-27:
Repetier does not have the same problem

Sebastianv650, 2018-01-27:
First thing that comes into mind reading this would be rounding. I might have seen something related to that on my extruder: Some day it noted that if I do a series of retract + primes moves very often in a loop (testest without filament loaded), the extruder isn't in the original position after that but slightly advanced (rotated in extrusion direction). I blamed my printed gregs wade extruder for that and wasn't thinking about it any more.

I'm very good in doing stepper.cpp dumps with speed or position values due to my planner and LIN_ADVANCE investigations, I will put it on my list..

Sebastianv650, 2018-01-27:
If, and I want to emphasize that if, that's realy a bug, it could be a quite old one as it also would explain some infill to perimeter artefacts I always have seen in prints with Marlin even in my 1.0 time.
On the other side, why we are seeing our prints not leaning to one side along z..?

tcm0116, 2018-01-27:
You might try increasing `MINIMUM_STEPPER_PULSE` to 1 or 2. It should be 1 for the A4988s and 2 for the 8825s based on their minimum step pulse width requirements.

GMagician, 2018-01-27:
@tcm0116 

> You might try increasing MINIMUM_STEPPER_PULSE to 1 or 2. It should be 1 for the A4988s and 2 for the 8825s based on their minimum step pulse width requirements.

that's interesting what is the purpose of this define? default is 0 why for 8825 it's better 2?

tcm0116, 2018-01-27:
A4988s can handle step pulse durations as low as 1 micro second but the
8825s need at least 1.9 us.

p3p, 2018-01-27:
The define indicates the minimum (guaranteed) time needed for a stepper driver to notice a step pulse in microseconds, the correct value can usually be found in the spec sheets. I think it is left at 0 by default as the MCU (8bit) usually can not run the stepper interrupts fast enough for it to cause issue, A4988s also do not appear to be so picky.

http://www.ti.com/lit/ds/symlink/drv8825.pdf Section 7.6

Namakemonoism, 2018-01-28:
MINIMUM_STEPPER_PULSE 1 or 2 gives the same results. 
The problem can also be recreated with 1/16 steps and 110 steps/mm and moving -0.1.
Something to do with the factors of 10?
I tried reversing motor polarity, but the problem did not reverse it continues to happen when moving -1.0/-0.1.
The difference in steps is around 75% not just nit picking 1 or 2 steps.

GMagician, 2018-01-28:
@p3p & @tcm0116 thanks, helpful info

psavva, 2018-01-29:
Is it possible this of a mechanical issue?  Check that there is no slippage on the toothed pulleys...

Does this happen only on the Y axis, of the X and Z axis as well?

Namakemonoism, 2018-01-29:
@psavva  it would be hard to consider this a hardware issue. 

2 brands of motors were tested using both A4988 and 8825 drivers at various voltages between 0.3v and 1.0v.
Testing was done with only a stepper, no addional pullies, belts or axis assembly was attached. Just a stepper with a piece of tape to make the rotation easy to see.

It happens in both X and Y. I would assume also Z if I tried the same steps/mm.

on full steps 11step/mm is no good but 10 is no problem.
on 1/16 steps 110step/mm is no good but 105, 120 were no problem.

Using Repetier I was able to run the same settings without trouble and calibrated manually, the positioning accuracy is within 20micron over a 50mm travel. This is with the motor turning a thick leadscrew, so I would say the motor has no torque or step skipping issues.

I program in Javascript and PHP, one time I programmed a sales calculator for our sales team. What @Sebastianv650 mentioned sounds quite possible. Finding a safe way to round and do math accurately to a bit of time. With 8 bit limitations there may be a maximum integer issue, but I am over stepping my knowledge beyond that and I dislike numbers in general.

Originally I thought I had made some mistakes in the config but I tried the same parameters in Repetiers config and it seems ok.

Sebastianv650, 2018-01-29:
> With 8 bit limitations there may be a maximum integer issue, but I am over stepping my knowledge beyond that and I dislike numbers in general.

I like numbers, especialy handy 8bit ones :-D Not so much room there to search for the missing bit..
I will do a test today. If I can reproduce it, it should be straight forward to find the code causing it..

Sebastianv650, 2018-01-29:
Hm, no signs of wrong behaviour. The planner is always reporting the exact number of needed steps at 10, 11, 105 and 110steps/mm. I also tested against the endstops, if the axis is moving at all it should hit or leave the endstop along time. It didn't. I tested with 1mm move commands back and forth, repeated 1000x.

Does this happen to you also if you add a little break between the back and forth movements? If not, it could be resonance.

Namakemonoism, 2018-01-30:
@Sebastianv650 which version Marlin did you run? Even only 10 times should have been enough to see it.

I was running movements with noticeable break between, should be enough time to avoid resonating. Also it happened with the same pattern both by interface or and g-code. Sending by g code should have removed the possibility of Pronterface being the problem. Marlin 2 is running smooth on RAMPS 1.4 & 2560? I could try it and see. 

Did you check if my config was unreasonable?

I tried 1 step/mm on the weekend with full steps and the result was quite strange. 2 or 3 commands were ignored, then the motor moved, then again 2 or 3 ignored, then the next one moved.

Sebastianv650, 2018-01-30:
I tested with recent RCBugfix, but there was no change I'm aware of in a very long time that could have changed the basic step execution way of doing.
Are you able to add some code lines for debugging if I give you some informations?

> Did you check if my config was unreasonable?

Only on a first glance. Your X and Y max. speeds are extremely low.

Namakemonoism, 2018-02-01:
@Sebastianv650 
The X and Y are slow now cause I haven't figured out a good setting and  didn't want to skip steps when I had my lead attached.

Let me know the debug steps, I can probably do it on the weekend.

Sebastianv650, 2018-02-01:
You can insert `SERIAL_ECHOLN(block->steps[X_AXIS]);` at the end of `Planner::_buffer_steps` inside planner.cpp. Replace X_AXIS by the axis you want to look at. The number of steps reported over USB is the number of steps executed. If your stepper isn't moving according to that, it's not Marlins fault. If the reported numbers are wrong, we have to digg further.

thinkyhead, 2018-02-14:
Looking forward to the next status update.

boelle, 2019-02-20:
@Namakemonoism status ??

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-06 by shitcreek