MarlinFirmware/Marlin #9371: Anet A8 + LCD12864 (Anet) + BLTouch: wrong characters on splashscreen & Bootloop with offsets

FP2K-Minske, 2018-01-27:
Setup:
Anet A8 with the original Anet LCD12864 Display and a BLTouch.
With Skynet3D it works with the same settings, so they should work with Marlin, too.

### Bug Report 1
- Description: While booting I get garbage on the splashscreen. It looks like I got the wrong charset. In the menu everything works like it should.
- Expected behaviour: Something like "Marlin Vers. 1.1.8" on the splashscreen
- Actual behaviour: Something like "öäöASfjAölsad"
- Steps to reproduce:
  - Boot up?

### Bug Report 2
- Description: When I define the offsets for the probe the printer didn't boot. When I connect to it (USB), I see it is booting every second. 
- Steps to reproduce:
  - Boot up?

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1670389/Marlin.zip)

FP2K-Minske, 2018-01-27:
Aaaaand I solved it.

The problem was... the language. If you set it to "de" for german, you can't set the offset. And the splashscreen didn't work. The language work with the last skynet, so maybe you should recode the language.

thinkyhead, 2018-02-14:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

FP2K-Minske, 2018-02-18:
Can I use my Configuration-Files or should I change the settings of the ones from the bugfix?

thinkyhead, 2018-02-23:
Try the former, and if that is too daunting, proceed to the latter.

closed 2018-03-07 by thinkyhead