MarlinFirmware/Marlin #9372: DUE TMC2130 SoftSPI

sh4jonny, 2018-01-27:
Hi all! How to build bugfix-2.0.x branch with teemuatlut/TMC2130Stepper in soft spi mode for DUE board? and how to config soft spi pins?
thank you!!!!

sh4jonny, 2018-01-27:
Can i use it in HW SPI maybe ?

teemuatlut, 2018-01-27:
HW and SW SPI are working with DUE and latest library from git.
The options to configure the pins and enable SW SPI are coming with PR #9114.

sh4jonny, 2018-01-31:
Hi All! Excuse me pls...  @teemuatlut 
i make 
#define TMC_USE_SW_SPI
#define TMC_SW_MOSI 4
#define TMC_SW_MISO 6
#define TMC_SW_SCK 5

in my pins_RAMPS_DUO.h

and get Driver erron PRINTER HALTED on my LCD...
where i'm wrong ? thank you

teemuatlut, 2018-01-31:
I will need more information to help you.
What's the serial output?
Does it halt right after booting?
What fork are you running?

sh4jonny, 2018-02-01:
@teemuatlut thank you for reply. so look. i try to run tmc2130 on DUE with hacked RAMPS with your TMC2130Stepper lib. Marlin last bugfix-2.0.x And try to do this in Soft SPI mode. i read PR #9114 and try to enable SW SPI like this
#define TMC_USE_SW_SPI
#define TMC_SW_MOSI 4
#define TMC_SW_MISO 6
#define TMC_SW_SCK 5
and I get Driver error PRINTER HALTED on my LCD... Yes right after booting.

teemuatlut, 2018-02-01:
Are you running the upstream bugfix-2.0.x or are you running my LPC_TMC2130 fork?
The software SPI integration hasn't been merged into upstream yet.

sh4jonny, 2018-02-01:
"The software SPI integration hasn't been merged into upstream yet."
there is a answer!!! thank you! i understand now!!!

closed 2018-02-02 by sh4jonny

NeelD87, 2018-02-22:
@teemuatlut  is software SPI now available in bugfix 1.1 branch ? I have the drivers working well in hardware SPI but connected them over LCD adapter and had to disable SD card.