MarlinFirmware/Marlin #9374: Thermal Runaway Heatbed

Lyr3x, 2018-01-27:
Printed today the first time PETG-Filament and my printer is getting very early a heatbed thermal runaway. Heatbed is set to 85° for the first layers and than to 75°. Short time before the thermal runaway is kicking in i got a temp message in octoprint:

Recv:  T:219.90 /220.00 B:73.07 /75.00 @:62 B@:127

For those temperatures the heatbed is heating between 73 and 76 degrees. Never had a problem when printing PLA (60°).

Any advices?

revilor, 2018-01-28:
Without any information regarding your printer or configuration I can only suggest to increase THERMAL_PROTECTION_BED_PERIOD in Configuration_adv.h. And if you are using PID for the heated bed and have not done already, run PID autotune.

Lyr3x, 2018-01-28:
Sorry for the missing information, was in hurry.

While checking the firmware this morning i saw that i inserted my PID values but i forgot to enable the PID BED stuff..

closed 2018-01-28 by Lyr3x

elmar1, 2018-11-05:
What is a PID value??

AnHardt, 2018-11-05:
Its actually 3 values. [PID](https://en.wikipedia.org/wiki/PID_controller)