MarlinFirmware/Marlin #9376: Compilation error with #define NO_WORKSPACE_OFFSETS and Delta

brunogm0, 2018-01-28:
### Compile Error
When I compile with `#define NO_WORKSPACE_OFFSETS` I get an error:
```
update_software_endstops   not declared
```

i think is because of  Marlin_main.cpp:L1284

GMagician, 2018-01-28:
Which version are you using? 1.1.8 or bugfix 1.1.x?
Both bugfix (1.1.8 and 2.0x) compile with no problems.

Maybe your configuration may help

brunogm0, 2018-01-29:
1.1.8, also tried the bugfix;
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/1674259/Configuration_adv.zip)

closed 2018-02-01 by brunogm0

blueeagle69, 2018-08-13:
I'm not sure if I'm running the bugfix, but my Marlin is 1.1.8.
I just set #define NO_WORKSPACE_OFFSETS to use my Resumer 3D, and it compiled and uploaded fine to my printer.

PR #9387 (2018-02-02): [1.1.x] Fix compile error on delta when NO_WORKSPACE_OFFSETS
Fix #9376 issue