MarlinFirmware/Marlin #9377: [2.0.X] SERIAL_PORT_2

kevintong116, 2018-01-28:
Is the SERIAL_PORT_2 now working in 2.0?
I would like to connect my Touch screen (sth similar to mks tft)
I set the SERIAL_PORT_2 to 3 and I modified the pins_RAMPS.h so that nothing is using pin 14 and 15
```
#define Y_MIN_PIN          100
#define Y_MAX_PIN           3
```
But I still can't get it connected.
What can I do?

cjsoong, 2018-01-28:
@kevintong116 It should not support software simulation uart.
So it can not be defined on a non-UART PIN

Nutz95, 2018-01-28:
I could make it work with my MKS using #define SERIAL_PORT_2 -1.
I'm connected on the AUX 1 output of my ramps board.

tcm0116, 2018-01-29:
`SERIAL_PORT_2` currently does not work with the AVR processors. What control board are you using?

kevintong116, 2018-01-29:
@Nutz95 Yes, I know AUX 1 would work, even when I set SERIAL_PORT_2 to 3
But as I want to control my printer using octoprint, I would need to connect my screen using other serial

@tcm0116 I'm using mks gen 1.4.

thinkyhead, 2018-02-14:
Likely this will be back-ported to AVR "pretty soon."

boelle, 2019-02-20:
@thinkyhead should this not have the feature request label?

boelle, 2019-06-22:
missing label

InsanityAutomation, 2019-06-22:
AVR hal does have code for serial 2, so i believe this is completed and can be closed.

closed 2019-07-06 by shitcreek