MarlinFirmware/Marlin #9378: [2.0] NEOPIXEL Compilation error

TGMods, 2018-01-28:
I had a major breakdown with my Sainsmart 2-in-1 so I have finally installed my RE-ARM. As I was having to rewire my printer I thought it would be a good idea to install some lighting as I had some strips of NEOPIXELS doing nothing.

However when I tried to build the image I got the following error.

### Compile Error
When I compile with `NEOPIXEL_LED` I get an error:
```
Compiling .pioenvs\LPC1768\src\src\Marlin.o
Compiling .pioenvs\LPC1768\src\src\core\serial.o
In file included from Marlin\src\feature/leds/leds.h:33:0,
from Marlin\src\Marlin.cpp:74:
Marlin\src\feature/leds/neopixel.h:32:31: fatal error: Adafruit_NeoPixel.h: No such file or directory
 
**************************************************************************

* Looking for Adafruit_NeoPixel.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Adafruit_NeoPixel.h"
* Web  > http://platformio.org/lib/search?query=header:Adafruit_NeoPixel.h
*
**************************************************************************
 
#include <Adafruit_NeoPixel.h>
^
compilation terminated.
*** [.pioenvs\LPC1768\src\src\Marlin.o] Error 1
```
Can anyone tell me what I'm doing wrong?
I have attached my configuration.h and a screen shot showing the  contents of the '.piolibdeps' folder.

[Neopixel Compilation Error.zip](https://github.com/MarlinFirmware/Marlin/files/1671109/Neopixel.Compilation.Error.zip)

GMagician, 2018-01-28:
Sure to use latest bugfix 2.0? I tried to compile yours configuration, for mega2560, and compile without errors

p3p, 2018-01-28:
Unfortunately, because the LPC176x platform is not an Arduino platform, we have problems supporting some of the "extra" features that require Arduino libraries, as this is a fundamental issue it is not easy to fix generically. Looking at the NeoPixel github page even if the LPC176x had an Arduino Core, it looks like the library is quite MCU specific on top of that requirement.

TGMods, 2018-01-28:
Thanks for the info. I knew going 32bit would have problems, I guess I'm stuck with 12V RGB strip instead of Neopixels for the time being

closed 2018-01-28 by TGMods

viktorglekler, 2019-08-01:
just type the path to Adafruit_NeoPixel.h in Marlin\src\feature/leds/neopixel.h. It worked for me