MarlinFirmware/Marlin #9382: [2.0.x] TMC2130 RADDS DUE and Octoprint - no connection to USB serial ports

Industrial-Spiderman, 2018-01-28:
### Bug Report
- Description:  
I have set up the Due and RADDS to work with Marlin 2.0.x and the TMC2130 drivers. I used the build of 26th Jan as well as the latest TMC2130 Library 2.1.1. I get a compiler error. Using the TMC driver library 2.0.1 version I had no compiler issues. This is the library I also used on a Mega 2560/RAMPS build with success and Marlin 1.17 Stable. All good for that.

- Now the issue:
For Due, TMC2130, RADDS and Octoprint all together I cannot connect to the USB ports.
I have checked so far that when I do comment out #define have-tmc2130, then everything compiles fine and I have a serial connection with the programming port at 115200 baud and can access the controller via Octoprint and USB. 

I see RADDS solutions on the internet, but those are all non TMC2130. Hence I gather there is still an issue with this specific combination. I would appreciate a fix or in case I am just misguided a hint how I can get this combination up and running.

[Conf files.zip](https://github.com/MarlinFirmware/Marlin/files/1671597/Conf.files.zip)

teemuatlut, 2018-01-28:
Please post the actual error message.

Industrial-Spiderman, 2018-01-29:
Hello Teemualut,

here the error message with 2.1.1 TMC2130Stepper

D:\My Documents\My Arduino Projects\libraries\TMC2130Stepper\src\source\TMC2130Stepper.cpp: In member function 'void TMC2130Stepper::send2130(uint8_t, uint32_t*)':
D:\My Documents\My Arduino Projects\libraries\TMC2130Stepper\src\source\TMC2130Stepper.cpp:97:67: error: invalid conversion from 'int' to 'BitOrder' [-fpermissive]
SPI.beginTransaction(SPISettings(16000000/8, MSBFIRST, SPI_MODE3));
In file included from D:\My Documents\My Arduino projects\libraries\TMC2130Stepper\src\source\TMC2130Stepper.cpp:3:0:

C:\Users\hans-_000\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\libraries\SPI\src/SPI.h:48:2: error:   initializing argument 2 of 'SPISettings::SPISettings(uint32_t, BitOrder, uint8_t)' [-fpermissive] 
SPISettings(uint32_t clock, BitOrder bitOrder, uint8_t dataMode) {
exit status 1
Error compiling for board Arduino Due (Programming Port).

teemuatlut, 2018-01-29:
This has been fixed already but I haven't pushed the changes yet.
I'll make sure it's in the next release of the library.

Industrial-Spiderman, 2018-01-29:
Thanks.

Any idea when that will be. I sort of cannot get the Due, RADSS, TMC2130 to by any means, so I would really appreciate....

teemuatlut, 2018-01-29:
v2.1.4 released. Should fix the compile issue.

Industrial-Spiderman, 2018-01-30:
Hello,

I can confirm that with MK4duo and Marlin 2.0.x the compiler error has been removed.

Industrial-Spiderman, 2018-02-01:
Hello,

So the motors are turning. I have still the issue that the homing does not go properly and the motors seem to be running rough. I will do some checking on my pins as well as the homing functions in the config files. Will come back in case I cannot identify an answer.

Industrial-Spiderman, 2018-02-02:
All adjustments done and runs fine. Thanks for the fast update

closed 2018-02-17 by thinkyhead

wbarber69, 2018-03-04:
I have a similar issue with marlin 1.1.8 where i get usb timeout errors when trying to connect to an mks gen L board with 2130 spi

thinkyhead, 2018-03-04:
@wbarber69 — Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If you still see the bad behavior we should investigate further.

wbarber69, 2018-03-04:
10-4, I’ll give it a go.  I’ve now got simplify3d directly connected and printing ;( once it’s finished I’ll get my laptop back and give it a try.  They are extremely quiet.  I’m very impressed.  It was a little weird having to invert all my axes.   I wasn’t expecting that.

thinkyhead, 2018-03-04:
As an alternative to inverting the axes, you can also just reverse the plug. It may be that the plugs on TMC drivers are flipped compared to common pololus.

wbarber69, 2018-03-04:
My plugs are keyed but yeah I get you

wbarber69, 2018-03-04:
Sorry, last question... will using the software spi be a good idea?