MarlinFirmware/Marlin #9386: configuration creality 3D CR-10S 500

boubouexpress, 2018-01-29:
# {Configuration.h}

## //setting the PID//

//Creality 3D Hotend CR-10S 500
  #define  DEFAULT_Kp 22.04
  #define  DEFAULT_Ki 1.52
  #define  DEFAULT_Kd 79.90

//Creality 3D bed CR-10S 500
  #define  DEFAULT_bedKp 386.11
  #define  DEFAULT_bedKi 22.49
  #define  DEFAULT_bedKd 1657.27

## //setting of the original firmware data//

#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 93 }

#define DEFAULT_MAX_FEEDRATE          { 300, 300, 5, 25 }

#define DEFAULT_MAX_ACCELERATION      { 500, 300, 100, 5000 }

#define DEFAULT_ACCELERATION          300    
#define DEFAULT_RETRACT_ACCELERATION  500    
#define DEFAULT_TRAVEL_ACCELERATION   1000  

#define DEFAULT_XJERK                 10.0
#define DEFAULT_YJERK                 10.0
#define DEFAULT_ZJERK                  0.4
#define DEFAULT_EJERK                  5.0

#define X_BED_SIZE 500
#define Y_BED_SIZE 500

#define Z_MAX_POS 500

## //solve the heating problem//

# {Configuration_adv.h}

#define WATCH_TEMP_PERIOD 240 

#define WATCH_BED_TEMP_PERIOD 480

## //solves the SD drive detection problem//

# {pins_RAMPS.h}

#define SD_DETECT_PIN     -1

# option BLtouch

#define BLTOUCH
#define X_PROBE_OFFSET_FROM_EXTRUDER -42  
#define Y_PROBE_OFFSET_FROM_EXTRUDER -11  
#define Z_PROBE_OFFSET_FROM_EXTRUDER -1.1

#define MULTIPLE_PROBING 2

#define GRID_MAX_POINTS_X 7

 #define LEFT_PROBE_BED_POSITION 40
 #define RIGHT_PROBE_BED_POSITION 455
 #define FRONT_PROBE_BED_POSITION 30
 #define BACK_PROBE_BED_POSITION 470

#define AUTO_BED_LEVELING_BILINEAR

//#define LCD_BED_LEVELING

#define LEVEL_BED_CORNERS

#define Z_SAFE_HOMING

## //change position of the park//

 #define NOZZLE_PARK_POINT { (X_MIN_POS + 20), (Y_MIN_POS + 20), 20 }

thinkyhead, 2018-02-14:
My hands are in too much pain to convert this information into Configuration files. Would you mind doing it for me?

Evgeny-SPB, 2018-11-08:
Hello!

I've spend a few hours looking at example configuration of Creality CR-10S and source code, that is available online at Creality web site.

CR-10S and CR-10 5S, are two different printers and there should be two example configurations for them.

But Marlin version 1.1.x is closed for development. Should I create configuration file for Marlin 2.x or 1.1.x ?

thinkyhead, 2018-11-09:
Yes, they should be made for `bugfix-2.0.x`, using the current example configurations as the starting-point, and submitted as a PR to that branch.

boelle, 2019-02-20:
@boubouexpress @Evgeny-SPB as thinky said make a PR

and i think we can close this one

Evgeny-SPB, 2019-02-27:
@boelle you are correct. Pull request is already merged in bugfix -2.0.x

I also think we can close this issue

boelle, 2019-02-27:
only @boubouexpress and @thinkyhead  can close

but i think @boubouexpress  just created the issue and then left :-(

@thinkyhead will you swing the axe over this one?

closed 2019-03-03 by thinkyhead