MarlinFirmware/Marlin #9389: Clone BL-TOUCH doesn't work correct

maks2018, 2018-01-29:
Hello, I have a clone of BLTouch and RAMPS 1.4 (MKS BASE 1.5)
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1674756/Marlin.zip)
 at delta printer. 

When I run auto calibration (G29 or G33), printer starts to do it at the height of  Z236.5, but there is #define DELTA_HEIGHT 284.00 in the configuration.

M119 will report TRIGGERED when the clone is deployed, and OPEN when it is stowed.

Please, help me with the configuration.

GMagician, 2018-01-29:
Usually bltouch should always return open because triggered is just for 5ms (impossibler to read)..

Try reading #9320 if there is something useful

ExtremeSitting, 2018-01-29:
I don't know much about this application with the BLtouch let alone a clone, but have you tried commenting out the BLtouch definition. Based on the original way you had to get these things working, the servo definition and the BLtouch definition are redundant (and possibly conflicting) in this situation. I could be totally wrong here, but it seems like it would be worth a try.

Bob-the-Kuhn, 2018-02-01:
The setup for a clone is different than for a BLTouch.

If I remember correctly the clone configuration is:
```
#define Z_ENDSTOP_SERVO_NR 0   // Defaults to SERVO 0 connector.
#define Z_SERVO_ANGLES {10,90}  // Z Servo Deploy and Stow angles
//#define BLTOUCH
#define SERVO_DELAY { 750, 0, 0, 0 }
```

thinkyhead, 2018-02-14:
Closing in 1… 2…

boelle, 2019-02-20:
@thinkyhead what number have you reached ? :-P :-D

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-06 by shitcreek