MarlinFirmware/Marlin #9391: How and where to use gcodes scripts to run when the firmware is loaded?

italocjs, 2018-01-29:
Where on the marlin code should i insert some gcodes to run once the arduino starts?

i would like to send a few lines of gcodes to run once the 3d printer starts (and this cant be done with the auto0.g file on sdcard), but i dont know much about the programming language used,

Also, i'm probably on the wrong place, where should i ask this kind of questions?

empakoso, 2018-01-29:
I solved this by using Octo Print and used the OnConnect Gcode Scripts.

thinkyhead, 2018-02-14:
Add to the end of the `setup()` function:
```cpp
  enqueue_and_echo_commands_P(PSTR("M300 S440 P200\nM300 S660 P250"));
```
…for example.

closed 2018-02-14 by thinkyhead

empakoso, 2018-02-14:
Cool! One to remember!! Thanks.

italocjs, 2018-02-24:
Thanks @thinkyhead  (=
It solved the problem here.

Also, for anyone who wants to use this, the setup() is inside the main_marlin.cpp

AnHardt, 2018-02-24:
'G','M' and 'T' are not used for parameter names, to not confuse the parser with the 'G','M','T' commands.
I have not looked up if the new parser is still prone to this - the old one was.

thinkyhead, 2018-02-25:
The parser itself is very robust now and will accept any letter as a parameter.
`T` is okay as a parameter. We've overloaded it without issue. Repetier Host still doesn't like it when you use `G` or `M` — always assumes they are commands. We avoid `N` also.

italocjs, 2019-12-28:
Sorry for reviving this defunct, but i didnt manage to get this command to work on 2.0.x

`enqueue_and_echo_commands(PSTR("M117 FCP Triggered\nM300 S660 P250"));`

I have made a new feature that i'd like to write a text on the display (using m117) and make a beep. but vscode says"message": "identifier \"enqueue_and_echo_commands\" is undefined". Any ideas?

[image]

thinkyhead, 2020-07-03:
Nowadays you would use:
```cpp
queue.inject_P(PSTR("M117 FCP Triggered\nM300 S660 P250"));
```