MarlinFirmware/Marlin #9394: LCD 12864 RepRap Full Graphic Smart Controller shifted #5703

Nulreference, 2018-01-30:
possibly the same as #5703 but in 1.1.8 tried solutions suggested in forums and on here to no avail
slight line corruption comes and goes about every 20ish seconds in the "fan" gfx on the upper right
single line of dots    only in fan area. sifts about 20% of screen width each shift and shifted display is also approx 1/2 high(or low depending on direction)
I can get to to shift right instead of left adjusting but i cannot find a stable setting from 0 to 4 on each 

#define ST7920_DELAY_1 DELAY_0_NOP
#define ST7920_DELAY_2 DELAY_1_NOP
#define ST7920_DELAY_3 DELAY_2_NOP

I can get to to shift right instead of left
approximate time per shift 40~45 seconds (continues to shift given time by the same amount each shift)

thinkyhead, 2018-02-14:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

Bob-the-Kuhn, 2018-02-15:
If you still have problems when all three delays are set to DELAY_4_NOP then it's time to consider possible hardware issues.

The RepRapDiscount Full Graphic Smart Controller is rather finicky about it's timing and is susceptible to electrical interference.

Try keeping the two cables separated from each other and from other cables.

Try shorter cables.

thinkyhead, 2018-02-23:
This issue is being closed due to lack of activity. If you have solved the issue, please let us know how you solved it. If you haven't, please tell us what else you've tried in the meantime, and possibly this issue will be reopened.

closed 2018-02-23 by thinkyhead