MarlinFirmware/Marlin #9395: UBL question

sailor-1, 2018-01-30:
Hello everyone,
I am using UBL for a while now ant its very good!. But when i want to change the z-offset ( ex. different materials) i have to flash again the firmware with different z-offset. Is there any other way? 
Should i use G29 P6 for small changes?

Bergerac56, 2018-01-30:
Hello

You are not obliged to reflash your firmware. 
If you have enabled babystepping in configuration_adv.h (#define BABYSTEPPING) and activated the option #define DOUBLECLICK_FOR_Z_BABYSTEPPING, you can modify your Z_offset value from the main screen (In this case a graphic LCD). At main screen you just "long" click on the encoder.

If you modify the Z_offset value, it is not automatically stored in the EEPROM, so you will lose your modif at power reset. To overcome that, you have to store the value in eeprom by going to "Control/Store settings" or doing a m500. So the values used at startup will be the one stored.

sailor-1, 2018-01-30:
@Bergerac56 Thank you for the answer.
I have enabled both. But when i use UBL, even if i make a Z-babystep it does nothing. (When i enabled billinear it works fine) . Even if it did work in UBL, if i reset the machine, it will loose the babystep value and again nothing is solved.

Also i modify the Z_offset from the LCD and then immediately press Store settings, but the problem is that in order UBL to work i have the following commands, which resets M502 an it looses the Z-Offset i just saved from LCD.
(as in http://marlinfw.org/docs/gcode/G029-ubl.html)
M502          ; Reset settings to configuration defaults...
M500          ; ...and Save to EEPROM. Use this on a new install.
M501          ; Read back in the saved EEPROM.
M190 S65      ; Not required, but having the printer at temperature helps accuracy
M104 S210     ; Not required, but having the printer at temperature helps accuracy
G28           ; Home XYZ.
G29 P1        ; Do automated probing of the bed.

G29 P3 T      ; Repeat until all mesh points are filled in.
G29 T         ; View the Z compensation values.
G29 S1        ; Save UBL mesh points to EEPROM.
G29 F 10.0    ; Set Fade Height for correction at 10.0 mm.
G29 A         ; Activate the UBL System.
M500

Bergerac56, 2018-01-30:
Just do not perform M502 as it recalls the "default" values stored in firmware. 

What I do:
For a given firmware and when I really need to restart from a clean configuration, I do the following:
Step 0:
M502 / M500 / M501  So I reload the default basic values and... obviously the firmware value for Z_offset.

Before executing a new UBL mesh, I carefully adapt the Z_offset. (Babystepping). My routine:
(in gcode, it is easier but I have built "user commands" for that too)
Step 1:
G28
G29 D (be sure UBL is off)
G28 Z (Home Z to take in account the Z_Offset value)
G1 Z0 (position Z at 0. With perfect Z_offset value, the nozzle would just touch the bed)
Now I check that the nozzle just touch the bed. If yes: next step. If not: "babystep" Z_offset (Main screen, long press. The Z axe does not move when adapting, so you add or substract a small value and check again (G28 Z/ G1 Z0) (G28 Z takes the new value of Z_offset and homes Z/ G1 Z0 positions Z at the new 0 value. If you are improving the Z_offset, it will be nearer or just on the bed) Be very carreful as the nozzle could damage the bed if too big values !

When you are happy with your value,
step 2:
Use a gauge (sheet of paper= +- 0.1mm or another known element (I use a piece of metal of 0.3mm) and check if everything is still good:
G28 Z / G1 Z0.3 (with my metal piece). If OK goto step 3. If not: adapt Z_offset again.

Step 3:
M502 to store the values in eeprom. After that DO NOT reload failsafe values but load settings if needed.
Perform a UBL mesh build
When mesh OK: G29 Sx (store the mesh in the "x" eeprom place) / G29 Lx / G29 A (activates UBL) / M500

At next start, the correct value of Z_offset will be loaded, UBL will be active with the mesh stored in the "x" (your choosed EEPROM place) place.

GMagician, 2018-01-30:
If I'm not wrong to let babystepping work on z_offset you have also to enable BABYSTEP_ZPROBE_OFFSET.
Obviously this as nothing to do to the fact that it's not working with UBL but only with bilinear

sailor-1, 2018-01-30:
Thank you for the answers. I just removed M502, M500, M501 from the start of the gcode and the z-offset is working as i want to. 
Closing the issue

closed 2018-01-30 by sailor-1