MarlinFirmware/Marlin #9397: [2.0.x] MKS SBASE can't compile

Moorviper, 2018-01-30:
Hi i checked out the [2.0.x] branch installed platform.io and VS Code...

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
> Executing task: platformio run <

[Tue Jan 30 12:28:38 2018] Processing LPC1768 (build_unflags: -Wall; build_flags: !python Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py, -fmax-errors=500 -std=gnu++11, -DU8G_HAL_LINKS; src_build_flags: -Wall; lib_deps: CMSIS-LPC1768, U8glib-HAL; extra_scripts: Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py; board_f_cpu: 100000000L; platform: nxplpc; lib_extra_dirs: frameworks; lib_ldf_mode: off; src_filter: +<src/*> -<src/config>)
-----------------------------------------------------------------------------------------------------------------------------------------------
Warning! Library `{'requirements': None, 'name': 'CMSIS-LPC1768'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'CMSIS-LPC1768'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
CorePackageManager: Installing tool-scons @ >=3.30000.0,<4
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
tool-scons @ 3.30001.0 has been successfully installed!
Uninstalling tool-scons @ 3.20501.2:    [OK]
Verbose mode can be enabled via `-v, --verbose` option
System: 100MHz
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
Modes: Finder/off Compatibility/light
Collected 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <CMSIS-LPC1768> v0.0.0
|-- <U8glib-HAL> v0.4
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_AVR/HAL_AVR.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_AVR/HAL_spi_AVR.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_AVR/MarlinSerial.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_AVR/persistent_store_impl.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_AVR/servo_AVR.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_AVR/watchdog_AVR.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/EepromEmulation_Due.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/HAL_Due.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/HAL_spi_Due.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/HAL_timers_Due.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/InterruptVectors_Due.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/MarlinSerialUSB_Due.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/MarlinSerial_Due.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/Servo_Due.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/persistent_store_impl.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/u8g_com_HAL_DUE_st7920_sw_spi.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/usb/ctrl_access.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/usb/sd_mmc_spi_mem.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/usb/sysclk.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/usb/udc.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/usb/udi_cdc.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/usb/udi_composite_desc.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/usb/udi_msc.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/usb/uotghs_device_due.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/usb/usb_task.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/watchdog_Due.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL_LCD_I2C_routines.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL_LCD_pin_routines.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL_spi.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL_timers.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HardwareSerial.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/LPC1768_PWM.o
In file included from Marlin/src/HAL/HAL_LPC1768/HAL_spi.cpp:58:0:
Marlin/src/HAL/HAL_LPC1768/HAL_spi.cpp: In function 'uint8_t spiTransfer(uint8_t)':
Marlin/src/HAL/HAL_LPC1768/spi_pins.h:38:20: error: 'P0_15' was not declared in this scope
#define SCK_PIN  P0_15
^
Marlin/src/HAL/HAL_LPC1768/fastio.h:47:99: note: in definition of macro 'LPC_GPIO'
#define LPC_GPIO(port)          ((volatile LPC_GPIO_TypeDef *)(LPC_GPIO0_BASE + LPC_PORT_OFFSET * port))
^~~~
Marlin/src/HAL/HAL_LPC1768/fastio.h:58:40: note: in expansion of macro 'WRITE_PIN_SET'
#define WRITE_PIN(IO, v)        ((v) ? WRITE_PIN_SET(IO) : WRITE_PIN_CLR(IO))
^~~~~~~~~~~~~
[... 118 lines cut ...]
^~~~~
Marlin/src/HAL/HAL_LPC1768/HAL_spi.cpp:154:11: note: in expansion of macro 'SCK_PIN'
WRITE(SCK_PIN, LOW);
^~~~~~~
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/LPC1768_Servo.o
*** [.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL_spi.o] Error 1
========================================================== [ERROR] Took 7.33 seconds ==========================================================

================================================================== [SUMMARY] ==================================================================
Environment megaatmega2560              [SKIP]
Environment megaatmega1280              [SKIP]
Environment anet10                      [SKIP]
Environment DUE                         [SKIP]
Environment LPC1768                     [ERROR]
Environment LPC1768_debug_and_upload    [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment STM32F1                     [SKIP]
Environment teensy20                    [SKIP]
Environment teensy35                    [SKIP]
========================================================== [ERROR] Took 7.33 seconds ==========================================================
Der Terminalprozess wurde mit folgendem Exitcode beendet: 1

Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste.
```

tofifee, 2018-02-02:
Did you copy the example Configuration files for the mks sbase before compiling?

Moorviper, 2018-02-02:
Yes i used this:
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/Marlin/src/config/examples/Mks/Sbase

tofifee, 2018-02-02:
Comment out your selected bed leveling option `#define AUTO_BED_LEVELING...` and see if it compiles now.

Am just at the same step ;)

Moorviper, 2018-02-02:
had none uncommented :-(

tofifee, 2018-02-02:
hm...

Hopefully thinkyhead or Roxy can chime in.

Got it compiled. Just finding the lil bugger firmware.bin ist a spot of trouble ;)

Moorviper, 2018-02-09:
OK i checked it out clean again:
uncommented the second port, and set the first to -1

And now it compiles
[image]
[image]
[image]

MS1987, 2018-02-11:
@Moorviper ,I tried your modification, but have another error:
`In file included from Marlin\src\module\../inc/MarlinConfig.h:41:0,
from Marlin\src\module\configuration_store.h:26,
from Marlin\src\module\configuration_store.cpp:47:
Marlin\src\module\configuration_store.cpp: In static member function 'static void MarlinSettings::report(bool)':
Marlin\src\module\configuration_store.cpp:1866:69: error: 'port' was not declared in this scope
#define CONFIG_ECHO_START do{ if (!forReplay) SERIAL_ECHO_START_P(port); }while(0)
^
Marlin\src\module\../inc/../core/serial.h:107:64: note: in definition of macro 'SERIAL_ECHO_START_P'
#define SERIAL_ECHO_START_P(p)             (serialprintPGM_P(p,echomagic))
^
Marlin\src\module\configuration_store.cpp:1881:5: note: in expansion of macro 'CONFIG_ECHO_START'
CONFIG_ECHO_START;
^~~~~~~~~~~~~~~~~
Marlin\src\module\configuration_store.cpp:1892:26: error: 'port' was not declared in this scope
SERIAL_ECHOLNPGM_P(port, "  G21    ; Units in mm");
^
Marlin\src\module\../inc/../core/serial.h:103:69: note: in definition of macro 'SERIAL_PROTOCOLLNPGM_P'
#define SERIAL_PROTOCOLLNPGM_P(p,x)             (serialprintPGM_P(p,PSTR(x "\n")))
^
Marlin\src\module\configuration_store.cpp:1892:7: note: in expansion of macro 'SERIAL_ECHOLNPGM_P'
SERIAL_ECHOLNPGM_P(port, "  G21    ; Units in mm");
^~~~~~~~~~~~~~~~~~
*** [.pioenvs\LPC1768\src\src\module\configuration_store.o] Error 1`

It seems error of port, have you sufferred it?

tofifee, 2018-02-11:
ateyou using anything other than -1?

MS1987, 2018-02-11:
@tofifee I have checked that it occurs after the commit of #8148. The previous version works well.

Moorviper, 2018-02-11:
checked out today and it still works...

```
git clone https://github.com/MarlinFirmware/Marlin.git 2_0Marlin
cd 2_0Marlin/
git checkout bugfix-2.0.x
cd Marlin/
rm Configuration*
wget http://hessen.meshviewer.net/SBASE/Configuration.h
wget http://hessen.meshviewer.net/SBASE/Configuration_adv.h
cd ..
sudo platformio run -e LPC1768
```

```
====================================================================================================== [SUCCESS] Took 25.10 seconds ======================================================================================================

================================================================================================================ [SUMMARY] ================================================================================================================
Environment megaatmega2560          	[SKIP]
Environment megaatmega1280          	[SKIP]
Environment melzi                   	[SKIP]
Environment melzi_optiboot          	[SKIP]
Environment DUE                     	[SKIP]
Environment LPC1768                 	[SUCCESS]
Environment LPC1768_debug_and_upload	[SKIP]
Environment rambo                   	[SKIP]
Environment sanguino_atmega644p     	[SKIP]
Environment STM32F1                 	[SKIP]
Environment teensy20                	[SKIP]
Environment teensy35                	[SKIP]
====================================================================================================== [SUCCESS] Took 25.10 seconds ======================================================================================================
```

MS1987, 2018-02-12:
@Moorviper I check again, it works! Maybe I have mix the old version and the new one. Thanks.
And now I update the "firmware.bin" through the tf card. However, the pc cannot recognize the usb port to be use serial communication(When I use smoothie, it can.), but it can recognize it as Udisk. I wonder what's wrong with it?

closed 2018-02-12 by Moorviper