MarlinFirmware/Marlin #9398: Delta + clone BLTOUCH (as Z Servo Probe) doesn't use calibration data

maks2018, 2018-01-30:
Hello! I've configurated bltouch clone in firmware as Z Servo Probe. It began to work, G33 started to check the bed and showed results
```
    READ: .Height:282.09 Ex:-0.95 Ey:-1.25 Ez:+0.00 Radius:103.74
    READ: . Tx:-0.21 Ty:-0.02 Tz:+0.24
    READ: Save with M500 and/or copy to Configuration.h
```
I've saved the result at eeprom (M500) and started G29 Automatic Bed Leveling. After checking I've got:
```
     READ: Bilinear Leveling Grid:
     READ: 0 1 2 3 4 5 6
     READ: 0 +0.128 +0.029 -0.073 +0.459 +0.095 +0.593 +1.715
     READ: 1 -0.070 -0.012 -0.079 -0.211 -0.329 +1.139 +0.625
     READ: 2 -0.282 -0.133 -0.133 -0.175 -0.242 -0.250 -0.287
     READ: 3 -0.295 -0.220 -0.201 -0.263 -0.283 -0.283 -0.058
     READ: 4 -0.389 -0.348 -0.308 -0.134 -0.081 -0.063 -0.066
     READ: 5 -0.082 -0.071 -0.098 -0.248 -0.237 -0.195 -0.024
     READ: 6 +0.144 -0.022 -0.085 -0.099 -0.058 -0.227 -0.229
```
I've saved the result at eeprom (M500) and started manual checking of bed level. The difference at some points was more then 0.4mm. I've started the G29 again and results were like previous. Without using G28 I've started manual checking of bed level at different points and the result was more then 0.4mm.

What am I doing wrong?

maks2018, 2018-01-31:
somebody?

Bob-the-Kuhn, 2018-02-01:
The clone is working (head isn't crashing into the bed) but the results are inconsistent.  Correct?

On some systems the heaters and/or fans cause enough interference that the results are inconsistent.  Try enabling **PROBING_HEATERS_OFF** and/or **PROBING_FANS_OFF**

maks2018, 2018-02-01:
Thanks for your reply!
I've activated PROBING_HEATERS_OFF and PROBING_FANS_OFF

```
**M502**
**G33**
READ: Calibration OK                                rolling back.
READ: .Height:283.48  Ex:+0.00  Ey:-0.05  Ez:-0.07  Radius:103.61
READ: .               Tx:-0.15  Ty:+0.06  Tz:+0.10
READ: Save with M500 and/or copy to Configuration.h
**G28**
**G29**
READ: Bilinear Leveling Grid:
READ:       0      1      2      3      4      5      6
READ:  0 +0.089 +0.130 +0.002 +0.358 +0.049 +0.510 **+1.484** 
READ:  1 -0.065 +0.018 +0.007 -0.073 -0.184 **+1.064** +0.683
READ:  2 -0.152 -0.013 +0.046 +0.036 +0.006 -0.065 +0.007
READ:  3 -0.082 +0.016 +0.055 +0.046 +0.031 +0.025 +0.115
READ:  4 -0.050 +0.026 +0.062 +0.087 +0.121 +0.133 +0.137
READ:  5 +0.054 +0.026 -0.024 -0.055 -0.061 -0.009 +0.146
READ:  6 +0.094 +0.052 -0.029 -0.104 -0.138 -0.177 -0.067
```
(There is a glass on the bed and it can't be so crooked, but results are always the same.)

Then (without G28) I check if the altitude map G29 is using.
G1 X0 Y0 - bed level **Z-0.1**
G1 X50 Y0 - bed level **Z-0.1**
G1 X-50 Y0 - bed level **Z-0.3**
G1 X0 Y50 - bed level **Z-0.5**
G1 X0 Y-50 - bed level **Z+0.4** 

At the different points height of the bed is different. Does it mean that received altitude map is not used?

Bob-the-Kuhn, 2018-02-01:
A glass bed will always have ripples and tilt.  The 1mm jump in values is definitely suspect.  Are they repeatable?

I have some not so fond memories of trying to use a BLTouch.  I never did get decent results from the M48 repeatability test.  I ended up using G26 and the UBL mesh editing tools to dial in the printer.

@Roxy-3D - suggestions?

maks2018, 2018-02-02:
More then 1mm jumps are always at the same points. Unfortunately, I don't know coordinates to check them manually :(

my M48 results:
```
READ: Finished!
READ: Mean: 0.020764 Min: 0.009 Max: 0.025 Range: 0.016
READ: Standard Deviation: 0.005963

READ: Finished!
READ: Mean: 0.021764 Min: 0.013 Max: 0.031 Range: 0.018
READ: Standard Deviation: 0.005877

READ: Finished!
READ: Mean: 0.015764 Min: 0.009 Max: 0.023 Range: 0.014
READ: Standard Deviation: 0.004311
```

maks2018, 2018-02-09:
Can anyone help me?

Bob-the-Kuhn, 2018-02-09:
The M48 results look good.

You may have a bump in the corner of the bed.  Does it follow the glass if you rotate the glass 180 degrees?  Have you tried flipping the glass upside down?

Glass does bend.  Maybe moving the clips will help.

If the bump doesn't follow the glass then look for something binding or pulling like cables or filament.   Maybe even dirt or film on the rail(s).  You can also try running a separate cable to see if you're picking up electrical interference.

The ultimate test is to run G26.  If you can get a decent pattern with G26 then all is well.

I don't remember if you can edit the points in the bilinear mesh.  I know that you can if you use the UBL system.

maks2018, 2018-02-10:
I tried without glass
```
M502
G33

READ: Calibration OK                                rolling back.
READ: .Height:285.20  Ex:+0.00  Ey:-0.09  Ez:-0.11  Radius:103.80
READ: .               Tx:-0.27  Ty:+0.05  Tz:+0.22
READ: Save with M500 and/or copy to Configuration.h

M500
G28
G29

READ: Bilinear Leveling Grid:
READ:       0      1      2      3      4      5      6
READ:  0 +0.081 +0.103 +0.000 +0.329 -0.005 +0.413 **+1.370**
READ:  1 -0.055 +0.039 +0.012 -0.084 -0.182 **+1.007** +0.699
READ:  2 -0.123 -0.001 +0.055 +0.072 +0.085 -0.001 +0.049
READ:  3 -0.088 +0.000 +0.038 +0.045 +0.084 +0.062 +0.112
READ:  4 -0.079 -0.004 +0.021 +0.075 +0.114 +0.122 +0.115
READ:  5 +0.016 +0.008 -0.001 -0.047 -0.050 +0.019 +0.122
READ:  6 +0.064 +0.027 -0.034 -0.108 -0.095 -0.114 -0.027
```

maks2018, 2018-02-10:
I tried to print with G26 - it's terrible :(

thinkyhead, 2018-02-20:
`G28` disables leveling. Probably shouldn't do it just before `G26`. Better to put `M420 S1` there. Meanwhile, `M501` resets your settings to defaults, so you also don't want that there.

maks2018, 2018-02-21:
but, M501 not M502, M501 - Load all saved settings from EEPROM. After execution the M501 G26 does not work without G28. 
```
M502
G33
M500
G28
G29
M500
M420 S1
G26
```
Correctly?

thinkyhead, 2018-02-23:
Sorry, yes, you have the right order now.

maks2018, 2018-02-25:
I tried to print with new code - it's terrible to 
[image]
:(

thinkyhead, 2018-02-25:
Try it again using `PROBE_MANUALLY` and using the paper test.

maks2018, 2018-02-25:
I defined PROBE_MANUALLY.  Doing G29 the nozle stops for a moment at Z15 and at once return to the home (Z284). 

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1756170/Configuration.zip)

thinkyhead, 2018-02-25:
Try `PROBE_MANUALLY` some more and we'll see what's going on.

Follow this procedure:

- Download Marlin `bugfix-1.1.x` or `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. See how far you can get. With `PROBE_MANUALLY` you must either use the LCD (best) or issue `G29` and `G1` repeatedly during the procedure.
- Copy the log output into a **.TXT file** and attach it to your next reply.

From these logs we should hopefully get a better idea of what's going on with your machine and narrow this down to a single cause.

maks2018, 2018-02-25:
I've updated firmware to version bugfix-2.0.x. Here's the log with SERVO_PROBE, tomorrow I'll do it with PROBE_MANUALLY

[SERVO_PROBE.txt](https://github.com/MarlinFirmware/Marlin/files/1756479/SERVO_PROBE.txt)
[Marlin-bugfix-2.0.x-Config.zip](https://github.com/MarlinFirmware/Marlin/files/1756481/Marlin-bugfix-2.0.x-Config.zip)

maks2018, 2018-02-26:
Hi, I've tried PROBE_MANUALLY again.
Doing G29 the nozle stops for a moment at Z15 and at once return to the home (Z284).

[ERROR_PROBE_MANUALLY.txt](https://github.com/MarlinFirmware/Marlin/files/1759313/ERROR_PROBE_MANUALLY.txt)
[Marlin-bugfix-2.0.x-Config_PROBE_MANUALLY.zip](https://github.com/MarlinFirmware/Marlin/files/1759315/Marlin-bugfix-2.0.x-Config_PROBE_MANUALLY.zip)

thinkyhead, 2018-02-26:
Hmm… The second log reveals nothing about the move back to center. That is very strange. Do you also find this to be the same behavior if you initiate `PROBE_MANUALLY` homing from the LCD controller?

maks2018, 2018-03-02:
Up

AnHardt, 2018-03-02:
@maks2018 
Who are you? If seen this ugly "Up"s before.

maks2018, 2018-03-02:
@AnHardt thank you for your "help"

thinkyhead, 2018-03-05:
[image]

maks2018, 2018-03-06:
@thinkyhead That was sarcasm. The problem is not solved :(

thinkyhead, 2018-03-09:
We appreciate sarcasm and find it motivating. Unfortunately, we have very little to go on in terms of solving your issue. You might try various forms of bed leveling and see if one of them works better in the meantime.

closed 2019-07-06 by shitcreek