MarlinFirmware/Marlin #9402: [2.0.x] RRD Full Graphic LCD not working on LPC1769 @ 120 MHz

tcm0116, 2018-01-30:
@Bob-the-Kuhn I could use your help. I'm trying to get Marlin working on an LPC1769-based board running at 120 MHz. With the board running at 100 MHz, everything works normally. However, when I increase the speed to 120 MHz, the LCD gets garbled. I'm certain it has to do with the SPI timing, but I'm lost in the U8G library interface. I see that there are both software and hardware SPI options in the HAL_LPC1768_LCD_defines.h header, but I can't seem to determine how it is decided if software or hardware SPI will be used. Any ideas?

tcm0116, 2018-01-30:
I think I finally found where U8G is configured:

https://github.com/MarlinFirmware/Marlin/blob/6d4bc9a1f847d51047814583834c5a4cbfeb2ae6/Marlin/src/lcd/dogm/u8g_dev_st7920_128x64_HAL.cpp#L211

As such, it looks like we're always using software SPI for the ST7920 LCDs, regardless of the `LPC_SOFTWARE_SPI` setting.

I realize I need get the software SPI working with the MCU running at 120 MHz, but it seems like it might be worthwhile to give the option to utilize hardware SPI. Would it make sense to enable to use of hardware SPI if `SDSUPPORT` is disabled?

tcm0116, 2018-01-31:
I was able to figure out how to get it to use hardware SPI, which is working as expected. As such, I need to re-work the software SPI implementation to be compatible with the LPC1768 running at 100 MHz and the LPC1769 running at 120 MHz. Perhaps delay functions could be used instead of repeatedly setting the outputs to get more consistent operation across the two frequencies.

Roxy-3D, 2018-01-31:
What we need is an RTOS under Marlin on 32-bit platforms.   We could have a separate process that banged the bits out and even went to sleep between clock pulses.   Or even partially through the generation of a clock pulse.

tcm0116, 2018-01-31:
>We could have a separate process that banged the bits out and even went to sleep between clock pulses.

Context switches are expensive and time consuming, so it wouldn't really be possible to sleep between clock pulses and still get reasonable speed. With hardware-based I/O, it would be more feasible to sleep while the hardware is doing its thing, which could be a improvement in bandwidth.

tcm0116, 2018-01-31:
```
#define LPC_SOFTWARE_SPI  // Re-ARM board needs a software SPI because using the
                          // standard LCD adapter results in the LCD and the
                          // SD card sharing a single SPI when the RepRap Full
                          // Graphic Smart Controller is selected
```

I'm becoming increasingly more confused by this statement. Why would it matter if we're using the hardware or software to generate the SPI data stream? It would seem that the software SPI is only needed if we're using pins that do not have hardware SPI support.

tcm0116, 2018-01-31:
That makes sense since it looks like the LCD was setup by default to use
software SPI. It might be worthwhile to modify the U8G bindings to use
hardware SPI if the pins are the same rather than forcing the SD card to
use software SPI. I was able to get U8G to use the hardware SPI for the
LCD, so it should just be a matter of getting the processor conditions
setup properly.

p3p, 2018-01-31:
yes it does all work on hardware SPI when the HAL SPI interface is modified to allow it, its how I originally had the platform setup, (LCD at 4Mhz sdcard at 8Mhz hardware spi) but it introduced other complications and I haven't had the spare time to focus on creating the missing parts for the HAL, this has lead to many workarounds appearing in lpc1768 platform to compensate.

tcm0116, 2018-01-31:
@p3p I've been reading up on #9114, and I didn't realize that the LCD and SD card worked at different speeds and at different modes. As such, I think with the current implement, we could only use hardware SPI for either the LCD or the SD card, but software SPI if both are being used.

tcm0116, 2018-02-01:
I was able to get this working. See #9423 for details.

closed 2018-02-01 by tcm0116