MarlinFirmware/Marlin #9403: Developers: Help on acceleration meaning in kinematic machines

Sebastianv650, 2018-01-30:
Hi together,
I'm trying to figure out how to get the e axis acceleration for my LIN_ADVANCE PR. I want to have something kinematic compatible, so I digged into the steps involed from G1 input until the planner gets the converted values. Most parts are quite straight forward, but I need some help understanding CORE and DELTA acceleration meaning:

Inside the planner, Marlin calculates the length of a move (`block->millimeters` and `inverse_millimeters`) and it's acceleration (`block->acceleration_steps_per_s2`, `block->acceleration`)

For cartesian it's easy:
block->millimeters is the length of the move in mm in XYZ space.
block->acceleration_steps_per_s2 is the acceleration in steps/s² along leading axis.
block->acceleration is the acceleration in mm/s² along move in XYZ space.

For CORE machines, Marlin uses special ways to calculate block->millimeters, so I guess this value is still mm traveled for the print move in XYZ space? If so, I assume the statements from cartesian above are true also for CORE?

For DELTA I'm "lost in space": There is no special DELTA section, so the math for cartesian is used. But instead of delta x, y, z values we have now the delta movements of the three towers a, b and c. So what is  the actual meaning of block->millimeters?
As this also influences the acceleration calculation, I also have no clue what the acceleration is refered to for deltas.

Hope somebody can bring light into the dark. Maybe @thinkyhead can ping a kinematic guru?

Sebastianv650, 2018-01-30:
Small DELTA example, maybe it's then easier for others pointing me on my failure:

A DELTA 10mm move along Z would mean raise all towers by 10mm.
block->millimeters = SQRT(10²+10²+10²) = 17.32mm
inverse_millimeters = 1 / 17.32 = 0.0577..

During acceleration calculation this would be used (acceleration is 1000mm/s² and 100 steps /mm):
steps_per_mm = block->step_event_count * inverse_millimeters = 1000 * 0.0577.. = 57.735 steps/mm
accel = acceleration * steps_per_mm = 1000 * 57.735 = 57735 steps/s²

Lets assume no axis max_acceleration is violated, so we can skip that.

block->acceleration_steps_per_s2 = accel = 57735 steps/s²
block->acceleration = accel / steps_per_mm = 1000 mm/s²

For acceleration, the block->millimeters value is not relevant at all. So the acceleration along print move in XYZ space **seems** to be OK. But it isn't! To get 1000mm/s² in Z, all steppers would have to accelerate at 1000 * 100 = 100000 steps/s². So the executed move will be way too slow in acceleration..?

Sebastianv650, 2018-01-30:
Let me do a guess, and sorry for tripple posting!

- block->nominal_rate is calculated with the same wrong steps_per_mm value. Therefore within the trapezoid calculation the acceleration distance is correct - again failure compensation.
- The stepper ISR will use the wrong block->acceleration_rate number, resulting in wrong acceleration values.
- segment_time_us depends on inverse_secs, which both are wrong. Therefore the block_buffer_runtime_us is also off (way higher than real), meaning the LCD will trigger updates even if this means it will kill the block buffer. This results in errors like stuttering in #9355.

Please tell me I'm wrong. This would mean DELTA mechanics is completely broken!

tcm0116, 2018-01-31:
Checkout my PR #8778

Sebastianv650, 2018-01-31:
At least I'm not the first who recognised this. So core machines are ok but delta and scara are broken. That's a big, big one, we need this pr!
Without it
- acceleration is not even close to what it should be
- speed is way of from the gcode nominal rate
- junction speed between blocks is never connected

I consider exact motion as THE core function of a printers firmware. Therefore in my eyes, we need a fix or we have to clearly state support for deltas is currently broken in all marlin versions! Basically we are filling everything but position values with random data. ..

Ouch!

mylife4aiurr, 2018-01-31:
Wow. Such I strange issue. So basically Delta's "work" by magic now? I couldnt tell we even a problem, delta prints fine with this issue - I wouldn't even know.

Sebastianv650, 2018-01-31:
Not magic, but all numbers are off. If you measure the real print speed, you should get a nice difference. For example just rause Z by 10mm at a given feed rate, calculate the time it should need and compare to reality.
Hint: Slow motion videos and frame by frame watching and calculating times by time index and / or frame rate can be useful.

Sebastianv650, 2018-01-31:
Or to keep it simple, use low speeds:
Home printer
G1 Z50 F120

Should take 25s. If not, we are right.

tcm0116, 2018-01-31:
On Delta machines, the XYZ velocity, acceleration, and jerk settings apply to the linear motion of the carriages that are driven by the steppers, and not to the cartesian motion of the nozzle. 

To alleviate any confusion, #8778 renames XYZ to ABC in the LCD in the Control/Motion submenus. It would probably also make sense to have the corresponding G-codes (such as `M92`, `M205`, etc) take ABC as parameters instead of XYZ. 

The real issue comes into play when the planner is trying to determine how long the requested movement is in order to apply the desired feedrate. For cartesian, this is easily done by taking the root of sum of squares (RSS) of the requested delta in movement for the X, Y, and Z steppers. Currently on kinematic machines, the RSS is taken of requested delta in movement of the A, B, and C steppers, which does not represent the physical distance the nozzle is to be moved. It still represents an amount of movement, but not any sort of cartesian movement, so the feedrate limiting still sort of works. In my testing, I could see that the speed of the nozzle would change when printing large circles. 

#8778 attempts to fix this issue by telling the planner what the cartesian distance of the requested movement is. That value is already being calculated in order to segment the movements on cartesian machines, so it's easily passed to the planner. It also reduces the number of RSS operations to one per G-code segment from one per sub-segment of each G-code segment.

tcm0116, 2018-01-31:
@Sebastianv650 for your test, the printer would need to be moved to Z0 after homing since Deltas home to +Z, but that's a very simple test that should show the issue.

Sebastianv650, 2018-01-31:
> That's true for both.

Sorry, but no it's not :) If that would be the case and the speed and acceleration is applied to the steppers, at least the Z only move should have the exact speed. As I have shown in the calculation above, that's not the case.

Sebastianv650, 2018-01-31:
To complete my calc. example:

inverse_secs = fr_mm_s * inverse_millimeters = 10mm/s wanted feedrate * 0.0577.. = 0.577
block->nominal_rate = block->step_event_count * inverse_secs = 1000 * 0.577 = 577 steps/s

This results in a stepper **and** nozzle movement speed of 577steps/s / 100 steps/mm = 5.77mm/s. Not the 10mm/s we asked for..

tcm0116, 2018-01-31:
>the Z only move should have the exact speed

That was my original thought, but then it occurred to me that the RSS of a 10mm movement in Z on a Delta machine is not 10mm.

```
SQRT(SQUARE(50) + SQUARE(50) + SQUARE(50)) = 86.6
```

As such, the planner will think it needs to move 86.6 mm at 120mm/minute, which should take 43.3 seconds and not 25 seconds.

Sebastianv650, 2018-01-31:
Exactly. Every move is executed at a wrong rate and acceleration rate. Only the mm/s values used for trapezoid calc. are OK due to error compensation, which gives a wrong feeling that everything is OK. The result is way of.

tcm0116, 2018-01-31:
>10mm/s wanted feedrate * 0.0577.. = 0.577

25/43.3 = 0.577

The source of error is that `inverse_millimeters` is wrong because `block->millimetres` is wrong because it is calculated as the RSS of the movement of the carriages and not the nozzle.

My PR passes `millimetres` to the planner in the cartesian coordinate space, which fixes the error you're referring to.

Sebastianv650, 2018-01-31:
> My PR passes millimetres to the planner in the cartesian coordinate space, which fixes the error you're referring to.

Exactly (again). That's why I state that at the moment every move is wrong, whatever related to stepper speed along it's axis or revered to the nozzle. With your PR, speeds should apply to nozzle and be OK.

tcm0116, 2018-01-31:
>Every move is executed at a wrong rate and acceleration rate

The current XYZ velocity, acceleration, and jerk limits apply directly to the movement of the stepper and not the nozzle. The feedrate passed by G1 is in terms of the nozzle. This is why I changed XYZ in the Control/Motion menu to ABC on kinematic since those settings apply to the movement of the steppers and not the nozzle.

Sebastianv650, 2018-01-31:
Planner calculation side-by-side with as-is today code values and with @tcm0116 PR to show it's not only fixing speeds, but also acceleration.
Using: 10mm Z only move. All towers have 100steps/mm. Feedrate is 10mm/s. Acceleration (M204 one) is 1000mm/s².

As is:
block->millimeters = SQRT(10²+10²+10²) = 17.32mm
inverse_millimeters = 1 / 17.32 = 0.0577..
With PR:
block->millimeters = passed as 10mm
inverse_millimeters = 1 / 10 = 0.1

Acceleration calculation as is:
steps_per_mm = block->step_event_count * inverse_millimeters = 1000 * 0.0577.. = 57.735 steps/mm
accel = acceleration * steps_per_mm = 1000 * 57.735 = 57735 steps/s²
With PR:
steps_per_mm = block->step_event_count * inverse_millimeters = 1000 * 0.1 = 100 steps/mm
accel = acceleration * steps_per_mm = 1000 * 100 = 100000 steps/s²

Lets assume no axis max_acceleration is violated, so we can skip that. Max. axis acceleration is defined with M201 and checked along A, B, C axis always.

As is acceleration results:
block->acceleration_steps_per_s2 = accel = 57735 steps/s² <- wrong!
block->acceleration = accel / steps_per_mm = 1000 mm/s²
With PR:
block->acceleration_steps_per_s2 = accel = 100000 steps/s² <- perfect!
block->acceleration = accel / steps_per_mm = 1000 mm/s²

Speed as-is:
inverse_secs = fr_mm_s * inverse_millimeters = 10mm/s * 0.0577.. = 0.577
block->nominal_rate = block->step_event_count * inverse_secs = 1000 * 0.577 = 577 steps/s <- wrong!
With PR:
inverse_secs = fr_mm_s * inverse_millimeters = 10mm/s * 0.1 = 1
block->nominal_rate = block->step_event_count * inverse_secs = 1000 * 1 = 1000 steps/s <- perfect!

Same is true for more complicated X, Y moves, but it's most easiest shown with Z only moves.

M204 acceleration applies to the nozzle move
M201 acceleration sets the limits along each tower A, B, C.
Feed rate (G1 F...) applies to the nozzle.
M203 Max feed rates applies to each tower A, B, C.

tcm0116, 2018-01-31:
@Sebastianv650 Excellent write-up! Thanks for putting that together. 

I assume that the M205 XYZ jerk settings apply to each tower A, B, C as well. 

One thing that I was considering in my PR is an overall maximum feedrate parameter. On cartesian, It's basically computed as the SRR of the X and Y maximum feedrates. However, that doesn't really make much sense for deltas since the M203 max feedrates apply to the movement of the towers and not the nozzle. This is also important for the delta segmenting since it currently uses whatever feedrate is passed to G0/G1 to determine how many segments to create. There's nothing preventing someone from passing some huge feedrate resulting in a single segment for a long move, which could drag the nozzle across the bed or part.

Sebastianv650, 2018-02-01:
> I assume that the M205 XYZ jerk settings apply to each tower A, B, C as well.

The jerk section is one of the trickier to understand ones, but yes, it should be apply to the towers and be also OK with your PR.

> One thing that I was considering in my PR is an overall maximum feedrate parameter.

I don't feel we need such a limit. Per axis (=towers) limits are set to prevent an overspeed that could damage the printer. Everything else is in the hands of the user, and if this person tries speeds which will make printing impossible, it's his/her fault. But other may think different.

boelle, 2019-10-24:
@Sebastianv650 still working on this one?

boelle, 2019-10-24:
if its just talk back and forth maybe use discord ?

boelle, 2019-11-24:
this one has gone stale and debate could as well be held on discord

Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2019-11-24 by boelle