MarlinFirmware/Marlin #9406: Pin conflicts RAMPS 1.4/Re-ARM

Grogyan, 2018-01-30:
With the pace of supporting the TMC2130 and TMC2208, the fact that some of the pins used for the extruder SoftSPI, conflict with the use of the MAX31855, whereby the MAX3155 CS uses pin 66.

Conversely pins 65 and 66 are not connected under Re-ARM, so therefore both Trinamic drivers and MAX31855 CS cannot use them.
This leaves very few pins available for CS or Soft SPI use.

So for myself, hopes of using the Trinamic SPI drivers along side the MAX31855 are looking to be impossible to implement.  Not that I can compile for MAX31855 for the Re-ARM 2.0 anyway.

teemuatlut, 2018-01-31:
You can try the hardware SPI for TMC drivers. It's [been confirmed](https://github.com/MarlinFirmware/Marlin/issues/8334#issuecomment-357421791) to be working.

The CS pins on Re-Arm are on the ETH header.

welverin, 2018-02-01:
@teemuatlut  but the spi work only with 2130, 2208 don't use spi, usa uart connection, or i miss something ?

necris38, 2019-01-13:
Is it helping ?

https://github.com/MarlinFirmware/Marlin/issues/8334#issuecomment-357421791

Grogyan, 2019-01-13:
I have spent months trying to get a working CS pin working on Re-ARM.

Have been using one of the MAX end stops as an alternative.  However, there is just compile issues because of the hard declaration of the hardware SPI for Arduino boards only, the LP1768 do not have those same registers.

Possible that the software SPI be used instead of the Hardware SPI for the MAX31855, and similar devices, with these ARM chips?
As those have significantly higher clock speeds to handle SPI on any port pin.

necris38, 2019-01-13:
Ok, I'm new on 32 bit card and I just ordered a Re-arm with a RAMPS 1.6+ and 4 TMC2208. Does this mean that it is a non-functional association? Or simply that I will have to configure the stepper in standalone mode?

Grogyan, 2019-01-14:
@necris38 The TMC drivers are fully supported on Re-ARM as a separate library.

The OP is about the SPI for other devices that use SPI, such as the MAX31855, which I, and many others,  use

katanahaha, 2019-01-14:
@grogyan Could you please guide us to wire up tmc2208 to re-arm + ramps 1.4 which pin to which is optimal solution?

boelle, 2019-02-20:
@Grogyan has the conflicts been fixed in latest bugfix 2.0?

lookiamluke, 2019-02-20:
I too am trying to figure out where I can connect my TMC drivers via UART. Glad that Marlin is supported on RE-ARM but really wish we had a workaround for this

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by Grogyan