MarlinFirmware/Marlin #9408: LEFT_PROBE_BED_POSITION can't be reached by the Z probe

kaliboe, 2018-01-31:
hi, I am still learning marlin configuration
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1681282/Configuration.zip)

my firmware was working fine until I just changed the bedsize from 220 to 210 for X and Y

// The size of the print bed
#define X_BED_SIZE 210
#define Y_BED_SIZE 210

now when I verify my firmware I got the messages exit status 1
static assertion failed: BACK_PROBE_BED_POSITION can't be reached by the Z probe.

I know it's something to do with the probe but it's hard to understand it for the moment, sorry
anybody who's willing to help me out here, thanks

attach my configuration.h file in zip format

GMagician, 2018-01-31:
Since you also changed LEFT_PROBE_BED_POSITION, RIGHT_PROBE_BED_POSITION, FRONT_PROBE_BED_POSITION and BACK_PROBE_BED_POSITION values, default have relationship with BED_SIZE, once you change your BED_SIZE value you need also to update them.
Values in these definitions must be reachable by your probe then BED_SIZE and ?_PROBE_OFFSET_FROM_EXTRUDER must be considered

kaliboe, 2018-01-31:
Thanks GMagician for your answer
Ok, I am following you, seems normal if you change the printbed you need to change the probe settings.
what does this do? _PROBE_OFFSET_FROM_EXTRUDER sorry I am a noob
I found this in my configuration.h file

 *   Z Probe to nozzle (X,Y) offset, relative to (0, 0).
 *   X and Y offsets must be integers.
 *
 *   In the following example the X and Y offsets are both positive:
 *   #define X_PROBE_OFFSET_FROM_EXTRUDER 10
 *   #define Y_PROBE_OFFSET_FROM_EXTRUDER 10
 *
 *      +-- BACK ---+
 *      |           |
 *    L |    (+) P  | R <-- probe (20,20)
 *    E |           | I
 *    F | (-) N (+) | G <-- nozzle (10,10)
 *    T |           | H
 *      |    (-)    | T
 *      |           |
 *      O-- FRONT --+
 *    (0,0)
 */
#define X_PROBE_OFFSET_FROM_EXTRUDER -10  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -33  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below +above  [the nozzle]

GMagician, 2018-01-31:
Yes, there is a relationship (in default configuration is hidden by definitions that uses bad_size)
You do probing over the bed and if bed changes dimension also the "probable" area will change.
If bed is smaller maybe you may can't probe in old position and if bed is bigger, maybe you don't want to
probe near the edges (you can have screws, a glass etc)
Logical that the real limits in probe are related to:
  bed size
  machine limits
  probe position vs nozzle position

Usually your probe is far from nozzle how much? Here comes in help ?_PROBE_OFFSET_FROM_EXTRUDER

The text you reported help to define sign on values. Look to an hypothetical printer. You have the nozzle and in front of it you have your probe (bltouch?) if your probe is 10mm in front of nozzle and perfectly aligned in X you have to write in X_PROBE... 0 and Y_PROBE.... -10

kaliboe, 2018-01-31:
ok, I can follow you, the probe setting is correct, but I made the bed 10mm shorter so the probe will go of the bed like it is now.
I need to tell the printer to move shorter now on the Y axis
where to I place this info?
GMagician I appreciated your help very much!

GMagician, 2018-01-31:
Probably you have only to subtract 10 to RIGHT_PROBE_BED_POSITION

kaliboe, 2018-01-31:
It works now
because I made the print-bed 10 mm shorter I subtract 5 to the FRONT_PROBE_BED_POSITION and subtract 5 BACK_PROBE_BED_POSITION

but the reason why I shorter the print-bed is because my wipe of the nozzle is next to the bed, so I was thinking, let's make the smaller then the wipe will just come on the bed, seems I am wrong again :-(

this script is running on my S3D software
`G28 ; home all axes
G1 Z5 F3000 ; lift
G1 X20 Y2 F1500 ; avoid binder clips
G1 Z0.2 F3000 ; get ready to prime
G92 E0 ; reset extrusion distance
G1 X120 E10 F600 ; prime nozzle
G1 X150 F5000 ; quick wipe`

GMagician, 2018-01-31:
Since home position has not been changed, when you reduce bed size you reduce only front/right end.

If you need to move wiping in another place just change your X/Y positions in G1 commands.
Instead of X20 and Y2 bigger numbers can help

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-06 by shitcreek