MarlinFirmware/Marlin #9409: Z Probe to Z_MAX pin

jamesbit, 2018-01-31:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

cjsoong, 2018-01-31:
@jamesbit  
Try this...
// #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
##define Z_MIN_PROBE_ENDSTOP Z_MAX_PIN

jamesbit, 2018-01-31:
I tried but I get : 
error "You must enable either Z_MIN_PROBE_ENDSTOP or Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN to use a probe."

revilor, 2018-01-31:
The comment in Configuration.h claims that "By default Marlin assumes the Z-Max endstop pin." As SanityCheck tells us this is not correct, you have to define the pin to use. Therefore

#define Z_MIN_PROBE_ENDSTOP
#define Z_MIN_PROBE_PIN <Z_MAX_PIN>

should do the trick. Replace <Z_MAX_PIN> by whatever pin the z-max endstop uses on your printer board (19 for RAMPS).

jamesbit, 2018-01-31:
Thank you revilor !
I have no error.
I will try it in the next days...

closed 2018-02-14 by thinkyhead