MarlinFirmware/Marlin #9411: Inductive sensor bed leveling

Thesecky, 2018-01-31:
Hi. After 2 days of troubleshooting i am turning to you. I have chosen bilienear bed leveling for my inductive sensor, i followed a link on youtube in setting up the marlin. After i set everything up and figured my correct z nozzle offset, i wanted to get my first bed leveling done. I did g28, which homed my probe to the middle of the print bed, and then i wrote G29.  After that the leveling process started to work. But at the end, i didnt see my grid and the values that the leveling detected. Have i missed a step, or am i doing the things wrong.  THANKS!

psavva, 2018-01-31:
Try:
G28 #Home All Axis
G29 #Probe the bed
G29 S0 #/Produced the Mesh Report

Thesecky, 2018-02-01:
HI. I tried that and the mesh is there. But when i tried to print, i put that as my starting script.

G28 W ; home all without mesh bed level
G29 ; activate the Z compensation bed leveling
G92 E0 ; reset extrusion distance
G1 X0 Y20 Z0.2 F3000 ; get ready to prime
G1 X100 E20 F600 ; prime nozzle
G92 E0 ; reset extrusion distance
M205 X10 ; jerk x standard 15
M205 Y10 ; jerk y standard 15

At the beginin of print, the printer started to probe the bed as expected. But when it moved to the starting location- to prime the nozzle, the nozzle went under the bed, the probe crashed straight into the bed. Why did that happened. My z nozzle offsed came -3.4.  Is that to much, the difference between the nozzle and probe is 1mm.

revilor, 2018-02-02:
With the probe 1mm above the nozzle a z offset of -3.4 seems wrong to me. How did you come up with that value?

AnHardt, 2018-02-02:
Do you have inductive/capacitive probe?
Then the blue/yellow round surface of the probe is completely irrelevant. What matters is the invisible "trigger point", 1/2/4 mm below that surface.

thinkyhead, 2018-02-17:
> G29 S0 ; Produced the Mesh Report

Actually `M420 V` outputs the mesh.

closed 2018-02-17 by thinkyhead