MarlinFirmware/Marlin #9412: UBL offset

fr33dimension, 2018-01-31:
UBL offset on the bed.
Yesterday I flashed 1.1.8 and wanted to use my old UBL data from 1.1.4. It didn't work, since on 1.1.8 the UBL see that sensor postion as the nozzle position and the entire mesh is offset. In other words, my probe is -43mm from the nozzle (left) my x max is 194mm, the mesh ends at 151 and the z axis rises up.
I probe 4 points per xy and should have at about 8 points to probe manually after P1, but when running a P2, it sais P2 finished. Any suggestions?
[Configuration.docx](https://github.com/MarlinFirmware/Marlin/files/1682830/Configuration.docx)

thinkyhead, 2018-02-14:
> wanted to use my old UBL data from 1.1.4. It didn't work

The doctor replied, "Then don't do that. That'll be $100."

Start with a fresh `bugfix-1.1.x` and configure your probe and leveling options from scratch.

Roxy-3D, 2018-02-14:
@fr33dimension    I believe this issue is resolved in both of the current branches.   Please check it out on your machine when you get a chance.

> I probe 4 points per xy and should have at about 8 points to probe manually after P1, but when running a P2, it sais P2 finished. Any suggestions?

Are you saying you can't fill unprobed mesh points with G29 P2?   If so...   That needs to be fixed.

fr33dimension, 2018-02-20:
@Roxy-3D trying 1.1.8 again later today.

fr33dimension, 2018-02-21:
No luck, I tried 1.1.8 and 2.0.0 both ubl are not working or I am doing something wrong.
Did anything change from 1.1.4 in terms of probe offsets? When I run a G29 P1 than a P2 there are not
unprobed positions found, but there are, my probe cannot reach the entire bed

Roxy-3D, 2018-02-21:
> Did anything change from 1.1.4 in terms of probe offsets?

Not directly.  But there was a change from X_MIN_POS and X_MAX_POS to X_BED_SIZE.   (And for Y)    That did cause some issues.

But really you probably don't need to do a G29 P2.    Pretty much, G29 P3 is going to fill any unprobed locations with reasonable values.    And if they aren't perfect...  you can edit the filled mesh points pretty easily.

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-03-11 by fr33dimension