MarlinFirmware/Marlin #9414: Independent Axis

MarlonBrandoII, 2018-01-31:
I am looking to use one of the extruder stepper motor drivers to create a separate linear motion axis (together with its min-end-stop). 

Seen a successful implementation of it on a YouTube video. I think MarlinX2 firmware was intended to support such a functionality (Dual X or Y axis). However, on GitHub, that looks a few years old.

I understand it is possible to have Dual Z axes, each with their own homing end-stops. This is achieved by simply uncommenting some code on the Configuration(_adv).h files. However, I need such two axes to be independently drivable (i.e., not concurrently moving all the time).

Could someone more experienced with Marlin kindly point me in the right direction? 

Much appreciated.

Sebastianv650, 2018-01-31:
Can you describe how your additional axis should be used? Marlin (by design) can only move axis in a synchronous way. If you want to start and stop an axis while other should keep spinning, that's impossible.

MarlonBrandoII, 2018-01-31:
The functionality that I seek is similar to that described in:

http://www.okob.net/projects/reprap_x2v3

but perhaps even more general than that. Let's call the four independent axes X, Y, Z and W. Functionally, W should be identical to the other three axes: Steps/mm definable, a min-end-stop assignable, etc. Trying to create a kinematic device, and need the additional axis, that's all.

Sebastianv650, 2018-01-31:
This dual x thing is available. The trick is that there is always only one X axis moving, or both together in a copy mode. So each move is one move in 3D space. If you want to add another axis in space, you have to think about what the meaning of this axis should be. Keep in mind that it's not possible to create a smooth connected movement along for axis if some parameters are not kept. On our 3 axis + extruder machines the 3 axis are always linked, but the forth axis can have speed jumps for example if the extrusion width is changed during a line path.

When you can specify the type of linked movement W has to do and how the planner should calculate it, you can think about implementing it. There is no general approach.

psavva, 2018-01-31:
A PR currently  exists which adds a fourth kinematic axis to Marlin for the Hangprinter by @tobbelobb

Maybe have a look what is find there...

MarlonBrandoII, 2018-01-31:
Just checked out the Hangprinter. Kudos to @tobbelobb! Awesome work, great machine! Thanks for the link.

It appears that the Hangprinter achieves control of the steppers using I2C connection to separate stepper drivers. I was hoping to use one of the five already present on the Ramps 1.4 platform.

@Sebastianv650, I would love to simply go ahead and use one of the two extruder drivers to control the W axis. The only question is how to home that axis. Is it possible to assign one of the max-end-stops not being used as a min-end-stop for the W axis?

empakoso, 2018-01-31:
Along the same lines, can I simply copy the movements of my extruder stepper driver pulses?  I am thinking about making a type of filament puller/dancer system.  (My apologies if I am hi-jacking your thread)

tobbelobb, 2018-01-31:
Thanks, I hope you find it useful :)

The Hangprinter PR adds a fourth kinematic axis on the normal E1, E2, E3, or E4 driver output, depending on if you have 1, 2, 3, or 4 extruders configured. The I2C stuff is not related to the addition of the fourth axis.

MarlonBrandoII, 2018-02-01:
Awesome, thanks for the clarification, @tobbelobb. 

Quick question: Did you have to make a great deal of changes to Marlin to accommodate a fourth kinematic axis with min-end-stop support? Is there a simple set of steps to implement that axis? Just trying to make my own changes (as minimal as possible) to Marlin, while keeping it intact otherwise.

tobbelobb, 2018-02-01:
Yes, there are quite a few changes. The most time consuming one for me was this: Some arrays are refer to cartesian axes as expressed in gcode, eg `current_position` and `destination`. Other arrays refer to motors, eg `target`. Marlin makes no conceptual difference between "XYZ as in gcode" and "XYZ as in my three kinematic motors" by default. They are both indexed the same way, like
```C
    current_position[X_AXIS];
    current_position[Y_AXIS];
    current_position[Z_AXIS];
    target[X_AXIS];
    target[Y_AXIS];
    target[Z_AXIS];
```
It takes a few hours to wade through all the arrays (and also a few functions) who need this distinction in order to work with a fourth kinematic axis.

I have not enabled min-endstop on any of my axes.

DerAndere1, 2019-02-19:
I am currently trying to implement an option to specify the number of non-extruder axes for cartesian setups. You may want to read the related discussion https://github.com/MarlinFirmware/Marlin/issues/8519#issuecomment-346986901 and have a look at the various branches of my Marlin fork, e.g. https://github.com/DerAndere1/Marlin/tree/bf2_6axis_dev . Maybe you spot the part that is missing to make the additional axes and endstops work.

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-06 by shitcreek