MarlinFirmware/Marlin #9416: Marlin Firmware Error

Leonnsiae, 2018-02-01:
### Compile Error
When I compile with my Marlin file I always get this error. I'm not sure what to change. I'm very very new to this and I am just confused. I'm trying to setup the firmware for a Wnaho Duplicator i3 V2
```
Paste_the_error_text_here
```
Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\dogm_lcd_implementation.h:39:0,

                 from sketch\ultralcd.cpp:28:

sketch\dogm_font_data_marlin.h:12:25: fatal error: utility/u8g.h: No such file or directory

 #include <utility/u8g.h>

                         ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

AnHardt, 2018-02-01:
```
 * IMPORTANT: The U8glib library is required for Full Graphic Display!
 *            https://github.com/olikraus/U8glib_Arduino
 ```

Leonnsiae, 2018-02-01:
see that's wh I am confused. Because I already added that to libraries.

AnHardt, 2018-02-01:
Use a 'matching' combination.
Either an older u8glib, or a newer Marlin. You are at Marlin 1.0.x or older - that's at least 3 years old. Try 1.1.x.

Leonnsiae, 2018-02-01:
Ohh okay. Now it worked. But my LCD still isn't turning on

psavva, 2018-02-01:
Can you please attach your config files?

Leonnsiae, 2018-02-01:
How exactly do i do that? Can't seem to paste from clipboard or select files

psavva, 2018-02-02:
Zip it up, and drag/drop it into the comment window.

locutus40, 2018-02-23:
get rid of the ide uploader and download an older version see if that helps.... then redownload the u8glib delete the one you have and start all over from scratch

closed 2018-03-07 by thinkyhead