MarlinFirmware/Marlin #9417: M108 not working (emergency parser enabled)

filipgoc, 2018-02-01:
I wonder if this is a feature or a bug. What am I doing wrong? 

### (Potential) Bug Report
- Description:
    - I am running 1.1.8 (with UBL).
    - I have emergency_parser enabled. Keep_host_alive is also enabled. 
    - M0 does stop the printer. I am running RAMPS with Full graphic smart controller. 
    - However, M108 does NOT do anything. Not over USB, not over Octoprint, not when it's hardcoded in the slicer gcode (As a simple M0 \n M108 sequence.)
- Expected behaviour: The printer should unpause on M108
- Actual behaviour: The printer does not unpause. 
- Steps to reproduce:
  - Somehow send in M0
  - Somehow send in M108

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/1683945/configs.zip)

thinkyhead, 2018-02-14:
Duplicate of #9254

closed 2018-02-14 by thinkyhead

jeffkyjin, 2019-07-22:
> I confirm the Bug!
> We have bug in G27 and M125 too.
> 
> I reported the bugs in this post: #9254

No bug, it all works.

If you use the host in Prontrun, you should stop the gcode sending thread and then send M108. Or you send M108, the printer will not get it so doing nothing.
It takes me several days to figure out it, hope this help others.