MarlinFirmware/Marlin #9419: Does UBL G29 have an `R` flag to change probing density on the fly?

filipgoc, 2018-02-01:
The marlinfw.org note on G29 UBL states a `R` flag. 
> [R<int>] | Repeat count. (Default GRID_MAX_POINTS_X * GRID_MAX_POINTS_Y)

I am running 1.1.8 with UBL and a fixed probe. 

However, no matter what I run (`G29 P1 R3` or  `G29 P1 R5` ...), my printer probes with the _default_ grid_max_points value. 

Is this a bug? Or is the documentation wrong?

filipgoc, 2018-02-01:
Third option: Is my syntax wrong? (Or whatever else I'm doing? )

Roxy-3D, 2018-02-01:
G29 P1 is intended to populate as much of the mesh as possible.  It does not need an 'R' option.   It does support a 'C' option to 'Continue' the mesh generation for the case where the probing needs to be stopped.

Squid116, 2018-02-01:
@Roxy-3D is it possible to create a grid with something other than GRID_MAX_POINTS_X * GRID_MAX_POINTS_Y?

It could be useful to store different sized grids in your various slots?  i.e.

Perfect a 3*3 grid, save it in slot 1 and use it to print
Notice some flaws and start making a 7*7 grid and save it in slot 2
Print a few thing using your 3*3 grid
Go back to tuning your 7*7

and so on...

filipgoc, 2018-02-01:
Thanks. 

I also have a follow up: 

Why does the UBL documentation list the R flag? Is this just wrong and should we fix the docs?
Or is there another syntax? I tried G29 Pn and that didn't do a thing.

Roxy-3D, 2018-02-01:
> @Roxy-3D is it possible to create a grid with something other than GRID_MAX_POINTS_X * GRID_MAX_POINTS_Y?
> It could be useful to store different sized grids in your various slots? i.e.

This can be done...  Probably, you would still have to build the firmware with the maximum size that will be used.  But a smaller size mesh could be loaded.   Mostly, it is a question of how useful this will be and how many people want it done.     The question I have is why not build your mesh at the higher resolution and use that?

> Why does the UBL documentation list the R flag? Is this just wrong and should we fix the docs?

I'm not sure what documentation you are talking about.   But the documentation in the code is here:  It tells you what the options are for the G29 P1 command.
```
   *   P1    Phase 1    Invalidate entire Mesh and continue with automatic generation of the Mesh data using
   *                    the Z-Probe. Usually the probe can't reach all areas that the nozzle can reach. On
   *                    Cartesian printers, points within the X_PROBE_OFFSET_FROM_EXTRUDER and Y_PROBE_OFFSET_FROM_EXTRUDER
   *                    area cannot be automatically probed. For Delta printers the area in which DELTA_PROBEABLE_RADIUS
   *                    and DELTA_PRINTABLE_RADIUS do not overlap will not be automatically probed.
   *
   *                    Unreachable points will be handled in Phase 2 and Phase 3.
   *
   *                    Use 'C' to leave the previous mesh intact and automatically probe needed points. This allows you
   *                    to invalidate parts of the Mesh but still use Automatic Probing.
   *
   *                    The 'X' and 'Y' parameters prioritize where to try and measure points. If omitted, the current
   *                    probe position is used.
   *
   *                    Use 'T' (Topology) to generate a report of mesh generation.
   *
   *                    P1 will suspend Mesh generation if the controller button is held down. Note that you may need
   *                    to press and hold the switch for several seconds if moves are underway.
```

> Or is there another syntax? I also tried G29 Rn and that didn't do a thing.

There are several modifier parameters.  R (and, X, Y, C) are not top level commands.  They are modifiers for the G29 command you give.   So...   it is true that G29 R2 won't do much.   It doesn't have a top level command to modify.    And in fact....  a G29 won't do any thing either.   That is because no top level G29 command was given.

filipgoc, 2018-02-01:
I'm sorry @Roxy-3D, there's just something I'm not getting.

# 1.

The P1 section does not list R, but the page itself does. Both on the top in the list of flags, and in the table that describes the flags. 

I'm talking about this one http://marlinfw.org/docs/gcode/G029-ubl.html

[image]

Here's the code from https://github.com/MarlinFirmware/MarlinDocumentation/edit/master/_gcode/G029-ubl.md

```
tag: R
    optional: true
    description: Repeat count. (Default `GRID_MAX_POINTS_X * GRID_MAX_POINTS_Y`)
```
# 2. 
If it's not supposed to be used either as G29 P1 R3 nor as G29 R3 then how? And if not at all, then ... why is it listed in the documentation for UBL G29?

filipgoc, 2018-02-01:
> The question I have is why not build your mesh at the higher resolution and use that?

For me this would be useful for development. When I'm testing various UBL related functions and workflows, I may want to do a quick small grid, while for production I may want a bigger one. Of course, this only feeds laziness of not re-flashing the firmware all the time. Definitely a niche use.

Squid116, 2018-02-02:
> The question I have is why not build your mesh at the higher resolution and use that?

I have a 430mm * 430mm bed - creating a high resolution mesh (that works) takes a lot of time, even a G26 takes a fair chunk of time to print - then adjusting each point etc., to get to the final mesh.

If I could create a lower resolution grid that will do and allow me to print - it might not be perfect, but it will mean I can print.

It also means I could set the max to 15*15 which is excessive, and then define a lower one based on my usage, the specific bed etc.

Roxy-3D, 2018-02-02:
> The P1 section does not list R, but the page itself does. Both on the top in the list of flags, and in the table that describes the flags.

The 'R' parameter is for cases where the user wants to specify a number of points.   For example...  Suppose you have done a G29 P1 but there are unprobed points on the left and the front of the bed.   You could move the nozzle to the left front corner of the bed and say G29 P3 R4 C 0.00   This will tell the firmware you want to fill the 4 closest points to the nozzle with a value of 0.00    

> I have a 430mm * 430mm bed - creating a high resolution mesh (that works) takes a lot of time, even a G26 takes a fair chunk of time to print - then adjusting each point etc., to get to the final mesh.

I have a gMax 1.5+ printer and it has a similar sized bed.  I'm currently using a 10 x 10 mesh but I really should change it to 15 x 15.    I'm talking out of both sides of my mouth here.   I know I should change to a 15 x 15 mesh...  And I could get the bed probed and a G26 printed in 30 minutes.   But I'm too lazy to do it.   (And maybe another 10 minutes to use the G26 to get everything perfectly tuned.)    I know exactly what you are talking about.     But really...   an hour to get the bed leveling system perfectly setup up is not that much considering it will give you perfect 1st layers for many months.

Squid116, 2018-02-02:
@Roxy-3D It might take you an hour, but you live and breath this stuff!  The rest of us need (much) longer to work it all out! :)

filipgoc, 2018-02-02:
No Phase 1. Thanks. Got it.
I submitted a clarifying PR for the G29 UBL docs page. 

The confusion stemmed from the fact that Linear ABL does have such a function... the P flag.

filipgoc, 2018-02-02:
@Squid116 if OK with you, I'd close this one. If you really care about changing mesh density on the fly, maybe make it a separate Feature Request and link to this thread?

Squid116, 2018-02-03:
Yeah close it, it would be nice, but so wounld many other things :)

closed 2018-02-17 by thinkyhead