MarlinFirmware/Marlin #9420: v2.0.x bugfix  29-01-18 - Due/Radds "Error writing to EEPROM!"

bruce356, 2018-02-01:
Cartesian dual X Carriage printer, using "AUTO_BED_LEVELING_BILINEAR"  "GRID_MAX_POINTS_X 10"
Levelling data was saved successfully.
Needed to save "M218 T1 X0 Y0 Z0.0127" offset data but failed with the following:-

Paste_the_error_text_here
SENDING:M501
echo:EEPROM version mismatch (EEPROM=? Marlin=V50)
echo:Hardcoded Default Settings Loaded
>>>M218 T1 X0 Y0 Z0.0127
SENDING:M218 T1 X0 Y0 Z0.0127
echo:Hotend offsets: 0.00,0.00,0.00 0.00,0.00,0.01
>>>M500
SENDING:M500
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
start
echo: Watchdog Reset
Marlin bugfix-2.0.x
echo: Last Updated: 2018-01-20 | Author: (none, default config)
echo:Compiled: Feb  1 2018
echo: Free Memory: 83312  PlannerBufferBytes: 1408
echo:EEPROM version mismatch (EEPROM=? Marlin=V50)
echo:Hardcoded Default Settings Loaded
echo:SD card ok
------------------------------------------------------------
Could someone advise why I am receiving this error message - thanks
regards - bruce

cjsoong, 2018-02-01:
@bruce356 
First do initialize EEPROM or Run M503 cmd.

bruce356, 2018-02-01:
Thanks cjsoong, that worked
regards

closed 2018-02-01 by bruce356