MarlinFirmware/Marlin #9421: X and Y axes switched

classbproject, 2018-02-01:
I just configured and updated my MKS 1.4 board from 1.1.6 to 1.1.8 and I see that my X and Y axes are switched with each other. When I try and home the X axis, the machine moves in the Y-direction and vice-versa. How do I fix this? I have not made any hardware changes.

cjsoong, 2018-02-01:
@classbproject 
Maybe board type & Mechanical setting is wrong.
Upload your configuration.h file

classbproject, 2018-02-01:
Sure. I've attached the files for both the 1.1.6 and the 1.1.8. While upgrading all the way from RC8 to 1.1.6 I never changed any hardware configurations and it worked great. 
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1686004/configuration.zip)

classbproject, 2018-02-02:
Switching the core kinematics from Core XY to Core YX seems to have solved this problem.

closed 2018-02-02 by classbproject