MarlinFirmware/Marlin #9422: Can't compile marlin with Arduino 1.8.5

LukeBoroughs, 2018-02-01:
Arduino: 1.8.5 (Windows 7), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

WARNING: Category '' in library EEPROM is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library SPI is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library SoftwareSerial is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library Wire is not valid. Setting to 'Uncategorized'
Warning: platform.txt from core 'Arduino AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
In file included from sketch\MarlinConfig.h:40:0,

                 from C:\Users\Luke\Desktop\3d files\ROBO Rework\Marlin-1.1.x\Marlin\Marlin.ino:31:

SanityCheck.h:36: error: #error "Marlin requires C++11 support (gcc >= 4.7, Arduino IDE >= 1.6.8). Please upgrade your toolchain."

   #error "Marlin requires C++11 support (gcc >= 4.7, Arduino IDE >= 1.6.8). Please upgrade your toolchain."

    ^

exit status 1
#error "Marlin requires C++11 support (gcc >= 4.7, Arduino IDE >= 1.6.8). Please upgrade your toolchain."

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

ikarisan, 2018-02-01:
Did you update all of your Arduino libraries to their newest version?
Maybe you should use the hourly builds of the Arduino IDE: [Link to Windows ZIP](https://www.arduino.cc/download.php?f=/arduino-nightly-windows.zip)

LukeBoroughs, 2018-02-01:
How do I make sure the IDE is pointed to the newest install directory and not accidently picking up an old avr?

Bob-the-Kuhn, 2018-02-01:
My suggestion is to delete all the Arduino directories and do a fresh install.  On my Windows 10 machine they are:
C:\Program Files (x86)\Arduino
C:\Users\bobku\AppData\Local\Arduino15

LukeBoroughs, 2018-02-01:
That worked. Thanks.

closed 2018-02-01 by LukeBoroughs

FindusD, 2019-01-21:
I have the same problem.
With Arduino 1.1 it's no problem to compile marlin. 
But not with 1.8.
I have this errors:
-------------------------------------------------------------------------------------------
Arduino: 1.8.8 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Build-Optionen wurden verändert, alles wird neu kompiliert
In file included from sketch\SdFile.h:27:0,

                 from sketch\cardreader.h:8,

                 from sketch\cardreader.cpp:2:

SdBaseFile.h:38:8: error: using typedef-name 'fpos_t' after 'struct'

 struct fpos_t {

        ^

In file included from sketch\Marlin.h:10:0,

                 from sketch\cardreader.cpp:1:

c:\program files (x86)\arduino\hardware\tools\avr\avr\include\stdio.h:950:33: note: 'fpos_t' has a previous declaration here

 __extension__ typedef long long fpos_t;

                                 ^

exit status 1
using typedef-name 'fpos_t' after 'struct'
-------------------------------------------------------------------------------------------
I have no idea what to do...
Before installing 1.8 I did clean all directories and registry. By hand and with ccleaner.
:(
I have tried it with different marlin versions, always the same result.