MarlinFirmware/Marlin #9424: G91 relative positioning strange behavior

Bftech, 2018-02-01:
[1.1.8 config.zip](https://github.com/MarlinFirmware/Marlin/files/1685276/1.1.8.config.zip)

Hi !

Printer : AM8, marlin 1.1.8, config based on the anet a8, print without issue with cura and other slicers (USB and SD card printing), i'm using the jog control of Simplify3D to move the carriage but it doesn't go where i want (same with cura's jog control).

### Bug Report
- Description: Each times it try to move in relative mode (G91), the printer go somewhere else
- Expected behaviour: in relative positioning (G91) sending "G1 X10 Y10" from X110 Y110 should move to X120 Y120
- Actual behaviour: go to X89 Y 121 (changing each times)
- Steps to reproduce:
  - home X and Y axis (or all axis)
  - G90
  - G1 X110 Y110 (<- go to X110 Y110 without issues)
  - G91 
  - G1 X10 Y10 (<- don't go to X120 Y120)

If i send "G1 X10" or "G1 X-10", the carriage goes to the left but if i send "G1 X100" it goes to right (but not 100mm). Similar behavior with other axis, respect software endstops.
I didn't had this problem on a previous version of marlin (1.1.5 i think but not sure) but since 1.1.7 at least, i can't control the printer correctly in G91.

Is it a problem with my config ? As i couldn't find similar problems the past 3 days i doubt it's a bug in Marlin but i scrolled through my config files 5 times and nothing seems related.

ikarisan, 2018-02-01:
#8874

ikarisan, 2018-02-01:
#8874

ikarisan, 2018-02-01:
Maybe related to #8874

Bftech, 2018-02-01:
Thanks, it seems related yes. I tried to reset home offsets (M206) by doing M206 X0 Y0 Z0 and now i can control my printer in relative mode (G91) with S3D jog control, everything move in the right direction and by the right amount. 
But when i tried to set the origins somewhere, using S3D's "Zero X/Y/Z" buttons (wich use G92 X0/Y0/Z0) (i'm using my printer as a plotter so i need to set the origins where i want my drawings to start), and then moving, the bug is still here and nothing correspond to what i'm sending. 
Exemple : 
- home X and Y
- G1 X80 Y150 (goes somewhere correctly)
- G92 X0 Y0 (set this position as X0 and Y0)
- try to move with S3D's jog controls : 
    - G91
    - G1 X-10 (**goes to the right more than 10mm**)
    - G90

So for now i can somehow work around this bug, but i will need M206, as my nozzle is not in the center of the bed when i do G1 X110 Y110 (my bed is 220*220)

thinkyhead, 2018-02-11:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

Bftech, 2018-02-11:
I flashed `bugfix-1.1.x` with the same config as above.
After trying what I've wrote before (and more) everything seems to work as it should ! Thanks for fixing the problem !

closed 2018-02-11 by Bftech