MarlinFirmware/Marlin #9426: [2.0.x] Due/Radds Large hotend Temperature fluctuations

bruce356, 2018-02-01:
Cartesian dual X Carriage printer, have done several Auto PID on both hotends and can not achieve steady temperature (set point is 245 deg.). Constantly fluctuates between 243.8 and 245.8 deg. Using E3d v6 hotends with 40 watt heaters.

Is this as good as it gets? because with Repetier loaded on the same machine temperature fluctuation are only a few 10ths of a degree, not 2 degrees. 

Not criticising Marlin as it is a far better firmware overall but just using Repetier as a reference, so I assume this temperature fluctuation can be improved upon in Marlin.
Regards - bruce

psavva, 2018-02-01:
Have you run PID Auto Tuning?
Have you saved the pid values in your config file?

bruce356, 2018-02-01:
@psavva "have done several Auto PID on both hotends" yes and saved to eeprom and checked to make sure the pids are still there after a reset.

thinkyhead, 2018-02-16:
If the automatic PID tuning isn't working terrifically well then you may try hand-tuning to see if you can get it closer to a stable point. Use `M301` while the heater is on and while observing the temperature graph. Try raising and lowering each value by a small amount to observe its effects. While this is essentially what `M303` does, you may find that you can get closer to a stable value by making these small tweaks.

thinkyhead, 2018-02-17:
You may get better results with a small reduction in the `PID_MAX` value. And also check out this forum posting: [How-to: Quickly hand-tune your PID values](http://forum.seemecnc.com/viewtopic.php?t=11440).

bruce356, 2018-02-23:
@thinkyhead  thanks for the reply, I will try to set it manually. Thanks for the link
Regards - bruce

thinkyhead, 2018-05-03:
I can only reiterate my previous replies.

closed 2018-05-03 by thinkyhead

guysoft, 2018-05-03:
I managed to hand tune in the end. Bit it felt a little like voodoo. Even though I know the math behind PID. It took quite a few good hours.
It seems to happen because auto-pid is not taking something in to account. 
The main thing that I did was increase D by a lot.

thinkyhead, 2018-05-04:
That seems like a reasonable derivative value. I'm not sure why `M303` wouldn't be able to tune it. The current algorithm is exactly as it has been since Marlin 1.0. There might be some issue with resistance or noise in your wiring. As mentioned, some have had better results by reducing `PID_MAX`, which seems to be better for some MOSFETs. Anyway, hand-tuning is a perfectly good workaround!