MarlinFirmware/Marlin #9427: LCD freezes when sdsupport is enabled

oddwick, 2018-02-01:
### Bug Report
- Description: ---
using:
AZSMZ 12864 controller
ramps 1.4
marlin 1.1.8
libraries:
u8glib 1.19.1
sd 1.2.2

i got everything wired and working properly and works like a champ until sdsupport is enabled.  i believe i have it narrowed down to a fw issue

- Expected behaviour: ---
working as expected

- Actual behaviour: ---
when sdsupport is enabled, the screen goes from boot screen to main screen and then freezes after the on the second flash of xyz.  if an sd card is inserted, of sddetect is inverted, then it remains frozen at boot screen.  the board is still responsive, and on the controller the encoder, kill button and buzzer still work.  serial reports sd init fail without a card, and sd ok with a card.  if i use an adaptor, it reports volume.init failed as expected.  

  the controller is wired directly to ramps and there is no change in behavior regardless of pin assignments or connections.  i even have it to where the only connections are vcc, gnd, miso, mosi, clk, lcd_cs, lcd_a0 and the behavior doesn't vary.  

- Steps tried: ---
2 different ramps boards
3 megas
2 azsmz controllers
2 different cables
shortened cable to < 10cm
used a smart adaptor board and wired directly to ramps
changed pin assignments 
used (most) stock VIKI2 pin assignments
ran out of ideas (not what i tried, but where i am!)

once upon a time, i had it working on a ramps board with an old version of marlin (1.0?) but i have scoured my computer and for the life of me i can't find the original configurations i used!!!!  so i do know at one point in time it did work properly...

including entire marlin dir because i have made some changes to the configs just to separate the azsmz from viki because of the differences in pins and contrast
```
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1686752/Marlin.zip)

thinkyhead, 2018-02-16:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

Bob-the-Kuhn, 2018-02-16:
The AZSMZ_12864 uses the same driver as the VIKI2.  

The fix is not yet in the code base.

See Issue #9589 for a file that should fix the problem for bugfix-1.1.x users. 

The 2.0.x fix will probably be posted tomorrow.

closed 2018-02-24 by thinkyhead