MarlinFirmware/Marlin #9428: [Question] Limits on retract acceleration

GMagician, 2018-02-01:
Not sure about what I read on code but it seems that retract acceleration is not limited. Value on configuration are used directly...correct or is my oversight?

Sebastianv650, 2018-02-01:
You mean retract acceleration is not checked against E axis max. acceleration? True, but as retract acceleration is only applicable to E, it makes no real sense to check that. In my oppinion we shouldn't spend cycles during runtime checking for user faults.

GMagician, 2018-02-01:
ok but DEFAULT_MAX_ACCELERATION has values also for E then I expected a check against this limit.

Sebastianv650, 2018-02-01:
This limit is checked for non retract (print and travel) moves. A check at compile time (retract accel <= DEFAULT_MAX_ACCELERATION of E axis) would cost nothing, but it would be also quite worthless as it can be changed anytime after compilation on LCD or gcode.

GMagician, 2018-02-01:
ok thanks for quick response

closed 2018-02-01 by GMagician