MarlinFirmware/Marlin #9429: Error  #define DEBUG_LEVELING_FEATURE is enabled

andylh, 2018-02-01:
------------------------------------------------------------
### Compile Error
When I compile with #define DEBUG_LEVELING_FEATURE in configuration.h get an error:

Arduino undefined reference 'ftostr43sign(float const&, char)'

Everything compiles fine is this is not enabled.

GMagician, 2018-02-01:
I tried my version and it compiles with no errors

GMagician, 2018-02-01:
but compiled with platformIO

paukstelis, 2018-02-04:
I have the same issue. Tried compiling with arduino 1.8.2 and 1.8.5. I noticed the same thing with bugfix 1.1.8

GMagician, 2018-02-04:
Maybe attaching configuration may help. Only enabling DEBUG_LEVELING_FEATURE will compile correctly

thinkyhead, 2018-02-06:
Patched.

closed 2018-02-06 by thinkyhead

JackBradach, 2018-04-30:
It looks like this same issue occurs with the head of the 2.0 bugfix branch.  There needs to be an || ENABLED(DEBUG_LEVELING_FEATURE) added at utility.cpp:51.  Should I open a new bug?

thinkyhead, 2018-04-30:
Thanks! It's now patched.

commit 123c6c5d (2018-02-06): Num-to-string functions for DEBUG_LEVELING_FEATURE

Fix #9429