MarlinFirmware/Marlin #9430: [FR] G26 pattern is needlessly slow and troublesome

chambm, 2018-02-01:
Thanks for maintaining this great firmware!

When I was using bilinear leveling I would never use G26 because it was so much faster and less error prone to use a test pattern like this: 
https://www.thingiverse.com/thing:2751421
[image]

But of course this pattern doesn't know anything about the mesh coordinates. But it prints without any retractions, which makes it faster and easier to print (i.e. if the nozzle offset is wrong the print doesn't turn into a huge mess of disconnected circles).

Is it feasible to add an option for G26 to print continuously without retractions? I'd like to use UBL because it allows more mesh points.

tcm0116, 2018-02-02:
It would also be great if G26 could be cancelled and if you could still use the LCD while it's running to do things like adjust the Z offset.

mylife4aiurr, 2018-02-02:
Cant you cancel a running G26 by holding the click wheel in? Mistaken?

Roxy-3D, 2018-02-02:
> Cant you cancel a running G26 by holding the click wheel in? Mistaken?

Yes.   As well as other long functions like G29 P1

I'm in the process of adding a 'F' ast option to G26 that will print straight lines all the way across the bed and no circles at the mesh points.   (I had to move the 'F' filament option to a new letter.)

chambm, 2018-02-02:
Yes we can cancel easily but we can't baby step. Baby step would be very useful for assisting the overall mesh level.

Fast mode sounds great! After posting this I thought about it and the (diagonal) lines will be much better for sure, but some infill-like parts along the path would also be helpful to easily see whether the lines are being squashed enough to close gaps. And it can still be done without retraction. Thanks for considering it!

Squid116, 2018-02-02:
@Roxy-3D  if available you could enable fwretract for g26? that way could use the printer's settings to set retraction.

chambm, 2018-02-02:
Allow Q to be set to 0 to disable retraction? No new letter needed, just documentation. :)

Roxy-3D, 2018-02-02:
> After posting this I thought about it and the (diagonal) lines will be much better for sure, but some infill-like parts along the path would also be helpful to easily see whether the lines are being squashed enough to close gaps. And it can still be done without retraction. 

Diagonal makes sense for a number of reasons.   I was already thinking about how (as an option) to rotate the mesh validation pattern by 45 degrees.   

> if available you could enable fwretract for g26? that way could use the printer's settings to set retraction.

> Allow Q to be set to 0 to disable retraction? No new letter needed, just documentation. :)

Yeah!!!   Of course.   At least it is documented in the code!!!!

Squid116, 2018-02-02:
@Roxy-3D yeah if the user has fwretraction enabled then the g26 pattern should use g10/g11 instead of manually specifying retractions.

Roxy-3D, 2018-02-02:
If the user has enabled firmware retraction...   My guess is they want it enabled during calibration type things as well as during the printing phases.     Because that is what we are talking about doing.

chambm, 2018-02-05:
When doing `Mesh Edit`, is it supposed to be the nozzle that moves to the mesh point, or the probe?
I got the list of mesh points reported by `G29 W1`:
```
MESH_X = [0.000,14.286,28.571,42.857,57.143,71.429,85.714,100.000,114.286,128.571,142.857,157.143,171.429,185.714,200.000];
MESH_Y = [30.000,42.429,54.857,67.286,79.714,92.143,104.571,117.000,129.429,141.857,154.286,166.714,179.143,191.571,204.000];
```
When I use `Mesh Edit` to go to the various points, when I go to [0,0] it goes to X0, Y0. When I go to [14,14] it goes to X200 Y204. Why isn't [0,0] in the mesh at X0 Y30 (my probe Y offset is set to 30)?

 I'm trying to find an ideal substitute for G26, because it's just infuriating for me to sit and watch it blob filament on my nozzle even with the nozzle at very close to the right height. I'm currently trying with a pattern like: 
[image]

Where each line passes through a diagonal of the mesh points. The pattern is 2 pairs of parallel lines for each diagonal, with a gap between the pair. Each line of the pair will be double the extrusion width, while the gap should be half the extrusion width. If the gap is bigger than that (or if there's a gap between the outer 2 lines, which should have no gap), then that part of the mesh is too high. If the gap is squished or the 2 pairs are joined, the mesh is too low. But this only works if I can use the Mesh edit tool precisely so that I know which point I'm editing. In other words, either the nozzle or the probe should be over the point(s) that I'm editing.

My relevant defines:
```
#define X_PROBE_OFFSET_FROM_EXTRUDER 0     // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 30     // Y offset: -front +behind [the nozzle]
#define X_BED_SIZE 200
#define Y_BED_SIZE 205
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define MESH_INSET 0
```

boelle, 2019-02-20:
@thinkyhead maybe add feature request label?

boelle, 2019-06-22:
missing label

qwewer0, 2021-01-01:
@chambm Could you test G26 on the latest bugfix-2.0.x, maybe with this https://github.com/MarlinFirmware/Marlin/issues/20615?

Sebazzz, 2021-02-24:
I've been trying to adapt G26 to draw filled circles instead of crosshairs, in an attempt to make it a good bit more useful:

[image]

I'm not sure if it is due to the arcing, but I'm running into what appears to be slight overextrusion.

zalexzperez, 2021-05-02:
Is there a way to increase the travel speed for the G26 pattern test? It's painfully slow... and filament starts to ooze between points.

qwewer0, 2021-05-02:
> Is there a way to increase the travel speed for the G26 pattern test? It's painfully slow... and filament starts to ooze between points.

Do you mean `G26_XY_FEEDRATE_TRAVEL`?