MarlinFirmware/Marlin #9431: Thermal runaway problem

Yurtkany, 2018-02-01:
Hi i have Tevo Tarantula and i flashed marlin firmware but this time i receive Thermal runaway error.
#1 try : Part cooling fan off , heater on , heated bed on its working ok.
#2 try : Part cooling fan on , heater on , heated bed on this time temp started falling.

READ: ok T:202.95 /215.00 B:54.97 /55.00 @:127 B@:45
READ: Error:Thermal Runaway, system stopped! Heater_ID: 0
READ: Error:Printer halted. kill() called!

How can i fix ?

AnHardt, 2018-02-02:
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration.h#L429-L448

https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration_adv.h#L50-L85

Yurtkany, 2018-02-02:
Thanks. Finally i found problem. Solution is simple i replaiced the fan.

closed 2018-02-02 by Yurtkany

alsrmurad, 2019-06-28:
#if ENABLED(THERMAL_PROTECTION_HOTENDS)
#define THERMAL_PROTECTION_PERIOD 100 (40) 
#define THERMAL_PROTECTION_HYSTERESIS 15 (4)

#define WATCH_TEMP_PERIOD 25 (16) 
#define WATCH_TEMP_INCREASE 6 (4) 
#endif

#if ENABLED(THERMAL_PROTECTION_BED)
#define THERMAL_PROTECTION_BED_PERIOD 100 (20) 
#define THERMAL_PROTECTION_BED_HYSTERESIS 15 (2)
#endif

change the setting  configuration_adv.h, in brackets the original values and without bracket new values. i hope this will help you.