MarlinFirmware/Marlin #9432: Merlin Problems

Leonnsiae, 2018-02-02:
I can't for the life of me get firmware uploaded. I've managed to get one uploaded but I don't think it's working. My LCD lights up and shows the regular printer setting screen then says "Killed. Printer Halted"
I've tried so many different configs. I can't seem to get it right. Can anyone help me out? I'm trying to set up RAMPS 1.4 with Arduino Mega 2560 for a Wanhao Duplicator i3 V2

Bob-the-Kuhn, 2018-02-02:
Please ZIP up your Configuration.h and Configuration_adv.h files and attach to your reply.  

We'll need these files to try to reproduce the problem.

Leonnsiae, 2018-02-02:
[Wanhao-I3-Plus-master.zip](https://github.com/MarlinFirmware/Marlin/files/1687720/Wanhao-I3-Plus-master.zip)
[Monoprice_1_1_6.zip](https://github.com/MarlinFirmware/Marlin/files/1687721/Monoprice_1_1_6.zip)
Here are two that I have that I can't for the life of me get to work. The monoprice one I kind of can but I don't think it's fully setup for the i3 v2. I'm also having the problem of when it's loaded on and I disconnect it from the PC and start the printer the LCD doesn't turn on .

Leonnsiae, 2018-02-02:
OKay. I got the second zip (Monoprice) uploaded and my screen is working when I start up my printer. I'm not sure if everything is right because it just blinks to the Marlin startup screen on and off.

thinkyhead, 2018-02-02:
Apologies, but we can't reasonably support 3rd-party forks of Marlin based on older releases. We suggest you contact your vendor. Unfortunately we don't have a Wanhao i3 configuration at this time, but ask around the appropriate forums and someone may be able to provide you a newer version of Marlin set up for your machine.

closed 2018-02-02 by thinkyhead

thinkyhead, 2018-02-02:
I see the usual Wanhao Duplicator 6 config uses MOTHERBOARD `BOARD_ULTIMAIN_2` but in your configuration it's set to `BOARD_RAMPS_14_EFB`. Again, if what you have is a Duplicator 6 then you may want to try the latest Marlin.