MarlinFirmware/Marlin #9433: Heating bed by relay is too slow

lmaddio, 2018-02-02:
Hi, I would like some help for improving my experience with a heatbed [Mk2](https://www.amazon.com/Heatbed-Hotbed-Square-Printer-Mendel/dp/B077MYRH6X/ref=sr_1_3?s=industrial&ie=UTF8&qid=1517544294&sr=1-3&keywords=heatbed+mk2+red) using a mecanic relay with 12v power supply (same power supply for Ramps as for heatbed) 
It takes too long to heat the bed, Its even summer here with 30 degrees. I wanna to heat it to 90 for printing ABS, I don't have problem with 40 for PLA.
I wanted to know if someone could give some help with the configuration of this settings so it may help me.

I have this MAX_BED_POWER 255, commented PIDTEMPBED
I know it exists BED_LIMIT_SWITCHING which enables Hysteresis, but I haven't found really good information about it

And I have change this for testing

#if DISABLED(PIDTEMPBED)
  #define BED_CHECK_INTERVAL 4000 
  #if ENABLED(BED_LIMIT_SWITCHING)
    #define BED_HYSTERESIS 2 T>target-BED_HYSTERESIS
  #endif
#endif

Thanks!

Bob-the-Kuhn, 2018-02-02:
I'm running a 300x300 bed with a large heat spreader and a 24V supply.  It takes about 10 minutes for it to get to 90C.

First check **BED_MAXTEMP** in **Configuration.h**.

Next plot the bed temperature vs. time.  Pronterface and Repetier Host have built in functions to do this.

Maybe there's too much air moving over the heat bed.

There could be a problem getting power to the heater.  Feel the relay, cables & connectors. The cables should be cool.  Connectors can be warm.  The relay should be cool.

Measure the voltage at the bed heater.  There should be no more than 1V drop from the power supply to the heater.

cjsoong, 2018-02-02:
@lmaddio 
P(w)=V*I
To speed up the temperature of the hot bed, you need to replace a large power thermal bed.
Also you can add insulation under the hot bed.
Reference this video:
https://www.youtube.com/watch?v=Il4WCnaBSXg

closed 2018-02-26 by thinkyhead