MarlinFirmware/Marlin #9436: incorrect z movement after ABL enabled

jackhuntingtonliu, 2018-02-02:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

jackhuntingtonliu, 2018-02-02:
Hi 

I have a Delta printer. I am using Marlin 1.1 firmware. This is my problem.

when I set up ABL with bilinear mode and save the mesh to the EEPROM, the ABL will not enable even if I run the M420 S1.
However, If I set my ABL in 3 Points mode or linear mode. The Z axis will adjust it height, only before or after X-Y move to the new position. 
I don't know what information I should post in order to solve this problem. So if you are interesting in solve this problem and need more information. Please tell me. if you have same problem, please share it with me.  Thank you

thinkyhead, 2018-02-17:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

boelle, 2019-02-20:
@jackhuntingtonliu Please test with the latest `bugfix-2.0.x`

closed 2019-07-06 by shitcreek