MarlinFirmware/Marlin #9438: Setting Z-Offset in LCD (control > motion > z-offset)

Nordphilz, 2018-02-02:
Yesterday I flashed Marlin 1.1.8 to my printer. Since yesterday I only used Skynet, where I could define my Z-Offset at the printer (without re-flashing) in Control -> Motion -> Z-Offset.

In Marlin I can't find this option. Can you tell me how to activate that menu?

I already checked this thread https://github.com/MarlinFirmware/Marlin/issues/6103, but unfortunately it didn't really help me. Changing my config to FIX_MOUNTED_PROBE gave me serveral new errors. 

Thanks

cjsoong, 2018-02-03:
@Nordphilz 
Try this modified version
[https://drive.google.com/open?id=1Z3Np3wZxujkyALnC5TTmZFrftsHhgle6](url)

insanity67, 2018-02-11:
Will this come back to the official version?

Nordphilz, 2018-02-12:
Thank you very much.

closed 2018-02-19 by thinkyhead