MarlinFirmware/Marlin #9440: Bl Touch clone blinking after start printing

rjhubner, 2018-02-02:
The title says everything.
I have a 3d Touch installed in my redone Cubex 3D printer. It works fine homing and auto bed leveling.
After a while printing it starts to blink (alarm). Doesn´t bother at all but if an alarm is triggered something is wrong I guess.

psavva, 2018-02-02:
Do you have an external 5v supply blocked up to the bltouch?
If there is not enough current, its possible the pin is triggering the alarm when it drops unexpectedly

rjhubner, 2018-02-06:
Thanks psavva.
I´m using the 3dTouch connected to the first servo connector on ramps 1.4 and installed a jumper on that pins close to reset button to send 5v to those servo connectors. I´ve checked with the voltimeter and the voltage is correct (5V).
The probe pin is not dropping (at least I cannot see it doing it) and I guess the current is not the problem since I have 2 power blocks of 12v 30A each feeding that machine.
Tonight Im going to create a new 3-way cable to discard the possibility of a broken wire who can probably cause any bad contact between 3dTouch and Ramps.
Keep you informed

rjhubner, 2018-02-08:
Well. I´ve changed the cabling that connect the 3dTouch to the ramps and the behavior still the same.
So, the problem could be configuration or the sensor itself.

psavva, 2018-02-08:
Try an exrenal 5v supply with at least 1amp rating just to be sure.

thinkyhead, 2018-02-17:
Maybe some kind of current leakage or RF interference. It's hard to say. If you send `M280 P0 S160` does the blinking stop?

locutus40, 2018-02-23:
make sure.. bl touch is activated  make sure the deploy is 90,10.. its just at the top of configuration h file for these.. at the bottom make sure servos  are activate.. and at the bottom 3 lines down youll see 300

add this 300,300,300 then it should be working with marlin if not and its still blinking make sure the trigger pins arent reversed ie signal into negative on the board etc... make sure the servo pins are the right ones negative on the outside positive middle and signal left.....

thinkyhead, 2018-02-23:
If you have `BLTOUCH` enabled, all the settings mentioned by @locutus40 will be applied for you, so there is no need to get into the weeds with servo settings.

locutus40, 2018-02-23:
so what your saying is if you uncomment bl touch in config.h at the begning  even though servo settings on the bottom are // hashed out just leave em hashed out?

ive always enabled them... just in case

thinkyhead, 2018-02-23:
@locutus40 — Where have you been, man? Quiet for ages, and suddenly you're everywhere I look!
```cpp
#if ENABLED(BLTOUCH)
  #ifndef Z_ENDSTOP_SERVO_NR
    #define Z_ENDSTOP_SERVO_NR 0
  #endif
  #ifndef NUM_SERVOS
    #define NUM_SERVOS (Z_ENDSTOP_SERVO_NR + 1)
  #endif
  #undef DEACTIVATE_SERVOS_AFTER_MOVE
  #if NUM_SERVOS == 1
    #undef SERVO_DELAY
    #define SERVO_DELAY { 50 }
  #endif
  #ifndef BLTOUCH_DELAY
    #define BLTOUCH_DELAY 375
  #endif
  #undef Z_SERVO_ANGLES
  #define Z_SERVO_ANGLES { BLTOUCH_DEPLOY, BLTOUCH_STOW }

  #define BLTOUCH_DEPLOY    10
  #define BLTOUCH_STOW      90
  #define BLTOUCH_SELFTEST 120
  #define BLTOUCH_RESET    160
  #define _TEST_BLTOUCH(P) (READ(P##_PIN) != P##_ENDSTOP_INVERTING)

  // Always disable probe pin inverting for BLTouch
  #undef Z_MIN_PROBE_ENDSTOP_INVERTING
  #define Z_MIN_PROBE_ENDSTOP_INVERTING false

  #if ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
    #undef Z_MIN_ENDSTOP_INVERTING
    #define Z_MIN_ENDSTOP_INVERTING Z_MIN_PROBE_ENDSTOP_INVERTING
    #define TEST_BLTOUCH() _TEST_BLTOUCH(Z_MIN)
  #else
    #define TEST_BLTOUCH() _TEST_BLTOUCH(Z_MIN_PROBE)
  #endif
#endif
```

rjhubner, 2018-02-26:
Thanks people for all the help. 
@psavva ... I´ve tried an external 5V block. Unfortunetly I had only 5V 1A to apply. It started to blink as soon as I´ve turned it on. Even so your line of thinking is right.

@locutus40 ... I´ve had the bltouch ENABLED in config.h.  Even having some doubt if the 3Dtouch could use the Bltouch configuration.
Also, I´ve checked all the code lines you sent here attached (thanks a lot) and they match perfectly.

@thinkyhead ... Oh yes! M280 P0 S160 makes it stop blinking, in fact turns the LED off. Same when I use the "Reset Bltouch" at the display menu.

rjhubner, 2018-02-26:
PROBLEM SOLVED!
After all the hints you gave me guys (thanks so much) I decided to hack that screew inside the 3Dtouch.
First I´ve turned it clockwise half turn... same thing... back to original position.
Then I´ve turned it CCW half turn... VOILÀ... It works perfectlly now.
It is printing for 3 days in a row and no problem.

Well... any case is a case... mine was to adjust the screew only

thinkyhead, 2018-02-26:
@rjhubner — Good news! Yes, that seems to be the common cause.

rjhubner, 2018-02-28:
Oh yes! As it didn´t come with a manual, I´ve suposed that screew was to adjust the height of the probe needle. Sounds like it adjust hall effect or something else.

closed 2018-03-01 by thinkyhead

agentsmith1612, 2018-09-23:
I had this problem too.
I have a Geeetech A30 but built a new controller Box with RAMPS and Marlin inside, because Smartto Firmware sucked. 
But the problem is the same. I used the 20 pol cable for the 3D touch/BL touch sensor also like in the original version. 
After a while of printing or only staying (with hotend heater turned on) the pin drops. 

I have read a lot in the internet some say, firmware issue, some solved it with turning the top screw of the sensor in CCW direction one half or one turn to adjust some eletronic hall effect or so.

But for me nothing solved the issue.

Then I found out when the hotend heater stays off the pin never drops, when it stays on the pin drops after some minutes.

My conclusion is that current for the heater is inducing some disrupance in the signal cable for the servo. Because the wires are together in one line. So signal cable for Servo is beside the heater cable. 

My solution is that after the print starts I disconnect physically the signal for the servo at the controller board. 
The printer is printing for two hours now without any pin drop.