MarlinFirmware/Marlin #9442: Short to ground even when drivers are disconnected

MarioStoilov, 2018-02-02:
I am getting
```
X driver error detected
overtemperature
short to ground (Coil A)
short to ground (Coil B)
```
Even when there is nothing but the USB connected to the motherboard (no LCD, no drivers, no nothing)

This seems to happen when I connect a cable to MISD pin on AUX3 (MKS gen 1.4 board). This happens even if the other end of the cable is not connected to anywhere (tried several cables, all not touching adjacent pins or me). 

Using Marlin 1.1.8. On older versions (1.1.4) this problem does not occur (probably because this feature is absent from there) 

Would really appreciate some help.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1690085/Marlin.zip)

MarioStoilov, 2018-02-02:
Just pulled out an MKS 1.4 **base** and I get the same result (but connecting to the EXP2 first pin.

teemuatlut, 2018-02-02:
Communication error due to a bad response from the driver.
Check wiring, pin definitions and post your M122 output (need TMC_DEBUG).

MarioStoilov, 2018-02-02:
Thanks, I will check out in a bit.

In the mean time:

1. Communication error due to a bad response from the driver. - this happens even when the drivers are disconnected not on the actual board)
2. Check wiring - like above, no wires are physically connected to the board, except for the USB
3. The only files I've changed from the Marlin 1.1.8 are the two I uploaded, all pin definitions (for board MKS_13) are unchanged
4. post your M122 output - Will do as soon as I reflash and test

teemuatlut, 2018-02-02:
There's still a response even if the drivers are not connected.

MarioStoilov, 2018-02-02:
Noted,

Here's the M122 output

```
20:18:08.524 : X	Y	Z	E0
20:18:08.524 : Enabled		false	false	false	false
20:18:08.524 : Set current	800	800	800	800
20:18:08.524 : RMS current	1436	1436	1436	1436
20:18:08.524 : MAX current	2025	2025	2025	2025
20:18:08.524 : Run current	25/31	25/31	25/31	25/31
20:18:08.524 : Hold current	7/31	7/31	7/31	7/31
20:18:08.524 : CS actual		0/31	0/31	0/31	0/31
20:18:08.524 : PWM scale	0	0	0	0
20:18:08.524 : vsense		0=.325	0=.325	0=.325	0=.325
20:18:08.524 : stealthChop	false	false	false	false
20:18:08.524 : msteps		256	256	256	256
20:18:08.524 : tstep		0	0	0	0
20:18:08.524 : pwm
20:18:08.524 : threshold		0	0	0	0
20:18:08.524 : [mm/s]		-	-	-	-
20:18:08.524 : OT prewarn	false	false	false	false
20:18:08.524 : OT prewarn has
20:18:08.524 : been triggered	false	false	false	false
20:18:08.524 : off time		0	0	0	0
20:18:08.524 : blank time	16	16	16	16
20:18:08.524 : hysterisis
20:18:08.524 : -end		-3	-3	-3	-3
20:18:08.524 : -start		1	1	1	1
20:18:08.524 : Stallguard thrs	8	8	0	0
20:18:08.524 : DRVSTATUS	X	Y	Z	E0
20:18:08.524 : stallguard
20:18:08.524 : sg_result		0	0	0	0
20:18:08.525 : fsactive
20:18:08.525 : stst
20:18:08.525 : olb
20:18:08.525 : ola
20:18:08.525 : s2gb
20:18:08.525 : s2ga
20:18:08.525 : otpw
20:18:08.525 : ot
20:18:08.525 : Driver registers:
20:18:08.525 : X = 0x00:00:00:00
20:18:08.525 : Y = 0x00:00:00:00
20:18:08.525 : Z = 0x00:00:00:00
20:18:08.525 : E0 = 0x00:00:00:00

```

teemuatlut, 2018-02-02:
Also make sure the drivers have 12V power.

MarioStoilov, 2018-02-02:
Well, just checked, they get their 12V (when they are connected, of course)

teemuatlut, 2018-02-02:
Drivers are by Watterott or FYSETC?

MarioStoilov, 2018-02-02:
FYSETC

teemuatlut, 2018-02-02:
You need to unsolder the SPI bridge close to the chip.
https://github.com/MarlinFirmware/Marlin/issues/8480#issuecomment-348859419

MarioStoilov, 2018-02-02:
I already did( but haven't bridged the other two bins in that comment). But I am having trouble understanding how when the driver is disconnected this error would still occur

MarioStoilov, 2018-02-02:
I just tried flashing without STOP_ON_ERROR. Everything turns on and looks as if there is no problem. The motors go back and forward, the current is being regulated but the sensorless homing is just not working. When I send the homing command to an axis, it immediately just decides that it's current position is its home. Even if I disconnect the pin that goes to the S on the corresponding axis, I just hear a brief struggle for the motor to start spinning and then it stops

Strange stuff.

Still, I do really want to have STOP_ON_ERROR. Is there a way to move the SDI pin on the motherboard?

teemuatlut, 2018-02-02:
In the bugfix-1.1.x branch there was software SPI option added. See the `configuration_adv.h` file.

MarioStoilov, 2018-02-02:
Well, this is weird. I downloaded latest and greatest from bugfix 1.1.x, reenabled STOP_ON_ERROR, didn't touch the SPI and there is no short to ground errors anymore. The motors are still turning. 

The only thing that is still "broken" is the sensorless homing. When I send the command for homing, the axis goes 1 or 2 mm **in the wrong direction** and decides it has homed.

MarioStoilov, 2018-02-03:
I found the problem.

It seems that disconnecting the driver is an error itself. I enabled TMC2130 only on X and fully connected it and saw no problems. Afterwards I enabled it on Y and got the error. After I fully connected Y (X still connected), the error stop occurring.

closed 2018-02-09 by MarioStoilov