MarlinFirmware/Marlin #9443: BIBO Touch Laser X Configuration not available in example_configurations

shaktee, 2018-02-02:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.
------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`
[FR] I have configuration files for BIBO Touch Laser X 3D printer based on Marlin. I would like to contribute that to the example_configurations.

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

shaktee, 2018-02-02:
Patch attached
[BIBO.patch.txt](https://github.com/MarlinFirmware/Marlin/files/1690412/BIBO.patch.txt)

shaktee, 2018-02-02:
I apologize. I realized the custom machine name in the previous patch was a bit too custom! Please find updated patch here

[BIBO.patch.txt](https://github.com/MarlinFirmware/Marlin/files/1690417/BIBO.patch.txt)

GMagician, 2018-02-04:
This configuration has been added (#9467 and #9469).
I think issue can be closed

closed 2018-02-04 by shaktee