MarlinFirmware/Marlin #9444: Anet A8 pins in Marlin

stumichaels, 2018-02-02:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

melvinisken, 2018-02-02:
Which pins do you mean exactly?

stumichaels, 2018-02-03:
Actually most pins.  I'll give an example: the SERVO0   pin is defined as 27.  Pin 27 on the ATMEGA1284P is AVCC.  Another example is the X_STOP_PIN which is defined as 18 but 18 is GND; the actual x-stop pin is 21.

melvinisken, 2018-02-03:
Be aware that the Arduino pin numbering is different than the chip pin numbering. For example, ground pins don’t have any number. Google for 1284p Arduino pinout. Or did you consider this already?

stumichaels, 2018-02-03:
I had assumed that the Arduino and chip pin numbering was the same.  That's why I was confused by the lack of correspondence.  It quickly became obvious that the numbering wasn't the same.  So, there must be a document or table that relates the two.  My question is what is the name of the file that contains said document or table and where can I find it and is it easy to read?

melvinisken, 2018-02-03:
It depends a little on the library (or better: hardware definition) you use. The ATmega1284p is not officially supported as far as I know, so you need to install those (you did, if you were able to compile). I guess it was Sanguino. But anyhow, there is a hardware folder within the Arduino package, where you can find pins_arduino.h files for all supported controllers. The location depends on your OS. It shuold be a 10 second google to find out yours. Here's what my 5 second googling revealed:
[image]

stumichaels, 2018-02-03:
I found a similar display for the ATMEGA1284P in the 44 pin package.  So, the definitions are in pins_Arduino.h for your particular implementation.  I have my answer.  Thanks.  Now, the trick is understanding pins_Arduino.h.

[atmega1284-pintout]<https://eur02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F10966010%2F35770240-ecd5bd5c-0917-11e8-9ac7-02653e30b3bc.png&data=02%7C01%7C%7Ce6cf9288f90244acc80108d56b33899a%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636532791447426029&sdata=nZlUZfg2Rm%2FIJHS00rKORf1zlHo0pLdGNadnh78T67A%3D&reserved=0>

melvinisken, 2018-02-03:
Ok, fine. That's up to you then. You can ask at the Arduino forums in that case. I guess you can get better support there regarding Arduino stuff. I suggest to close this issue.

stumichaels, 2018-02-03:
Agreed, close the issue.  Thanks for your help.

melvinisken, 2018-02-06:
Just a note: you need to close it, since you opened it. Thanks.

closed 2018-02-06 by stumichaels