MarlinFirmware/Marlin #9447: PID autotune Heating Failed

Zacharandir, 2018-02-02:
### Bug Report
- Description: When using M303 to tune PID values I get a Thermal Runaway error.

- Expected behaviour: Sending M303 to the printer runs the tuning routine and returns new PID constants
- Actual behaviour: The printer halts with a thermal runaway error partway through the first cycle.
- Steps to reproduce:
  - Send M503
  - Wait.
  - Printer stops due to thermal runaway

This first started happening after i updated my firmware to a recent bugfix version. When I reverted to version 1.1.5 I had no issues, even though they have the same settings as far as i can tell. This issue has been observed on multiple machines after updating from Version 1.1.5 to bugfix1.1.x

Thank you for your help.

[Bugfix1.1.x..zip](https://github.com/MarlinFirmware/Marlin/files/1691102/Bugfix1.1.x.zip)
[Version 1.1.5.zip](https://github.com/MarlinFirmware/Marlin/files/1691103/Version.1.1.5.zip)

cjsoong, 2018-02-03:
@Zacharandir 
Try run M303 E0 S200 C8

Zacharandir, 2018-02-03:
Same outcome, this time it said heating failed

GMagician, 2018-02-03:
Maybe first things should be configure the machine in the same way (if you want same behaviours):
  WATCH_TEMP_INCREASE, MAX_BED_POWER are different (and may be related to your issue)

Zacharandir, 2018-02-03:
I changed those parameters and it still failed.  

Here is a graph of the temperature after  sending the command:
[image]

GMagician, 2018-02-03:
Sure are you using bugfix 1.1.x? because 1.1.8 has a bug on pid autotune, fixed in bugfix branch

Zacharandir, 2018-02-03:
That might be it I downloaded my copy mid-January, i'll see if it's fixed.

GMagician, 2018-02-03:
Fix done 14 days ago #9265

Zacharandir, 2018-02-04:
The latest version fixed it! thanks for the tip.

closed 2018-02-04 by Zacharandir